eclipse-embed-cdt/eclipse-plugins #434: Improve the build configurations

ilg-ul, 2020-11-08:
Due to historical reasons, the Eclipse projects used several build tools, initially as JDT only, later maven was added.

As of now, projects have multiple natures:

```xml
	<natures>
		<nature>org.eclipse.m2e.core.maven2Nature</nature>
		<nature>org.eclipse.pde.PluginNature</nature>
		<nature>org.eclipse.jdt.core.javanature</nature>
		<nature>org.eclipse.pde.api.tools.apiAnalysisNature</nature>
	</natures>
```

and

```xml
	<natures>
		<nature>org.eclipse.m2e.core.maven2Nature</nature>
		<nature>org.eclipse.pde.FeatureNature</nature>
	</natures>
```

For compatibility with maven the output folder defined in `.classpath` is

```xml
	<classpathentry kind="output" path="target/classes"/>
```

In a message posted on embed-cdt-dev list, Alexander mentioned:

> But do you have similar error when you compile this project in Eclipse IDE (not with Maven, but with PDE) using the updated target.
> I'm asking because Eclipse PDE can highlight some issues with .target or with dependencies.

Verify if the current configuration allows the builds that Alexander mentioned, and possibly improve them.

ilg-ul, 2020-12-05:
Alexander @ruspl-afed, is there any improvement to consider here?

ruspl-afed, 2020-12-08:
I would suggest to employ Eclipse Oomph, it is very helpful to automate workspace setup, perhaps next release cycle.

ilg-ul, 2020-12-08:
Eclipse Oomph?!

It looks like I have a new homework to do...

ruspl-afed, 2020-12-08:
no need :) I can try it when time permits, it should be similar with what we have in CDT