eclipse-embed-cdt/eclipse-plugins #442: Documents image trademark

jonahgraham, 2020-11-27:
The there is this image [image] replicated a few times in embed-cdt plugins. Do we have permission to use Acrobat logo here?

The icon appears in the Documents view (other places too?):

[image]

ilg-ul, 2020-11-27:
Do you have any suggestion how to address this?

jonahgraham, 2020-11-27:
Quick solution is to just create a simple PDF icon, or if there is suitably licensed one out there we can create a CQ to reuse it.

If you want I can handle this, there may be an icon in Eclipse icon repo already that is suitable.

ilg-ul, 2020-11-27:
Sure, if you want, please handle it.

jonahgraham, 2020-11-29:
Step 1 - here are the icons - converted from SVG to PNG with [eclipse image renderer](https://git.eclipse.org/r/plugins/gitiles/platform/eclipse.platform.images/+/refs/heads/master/org.eclipse.images.renderer/) SVG created with Inkscape based on a standard blank document (I used the document from [ant_buildfile.svg](https://git.eclipse.org/r/plugins/gitiles/platform/eclipse.platform.images/+/refs/heads/master/org.eclipse.images/eclipse-svg/org.eclipse.ant.ui/icons/full/obj16/ant_buildfile.svg))

[PDF of SVG + generated png](https://github.com/eclipse-embed-cdt/eclipse-plugins/files/5611545/pdficon.zip)

The Hi-DPI image looks ok, the low res one is less than perfect, but ok for a first try.

Step 2 - I will do a PR with these images replacing the existing ones later.

ilg-ul, 2020-11-29:
Ok, thank you.

jonahgraham, 2020-11-29:
> Step 2 - I will do a PR with these images replacing the existing ones later.

Done. See PR #447

ilg-ul, 2020-11-29:
Thank you, I merged it.

Regarding HI-DPI, I never tried the plug-ins on such display. Do they display properly?

jonahgraham, 2020-11-29:
My current setup is not hidpi, I have access to one if I need to. I just followed the standard practice of the @2x which automagically does the correct thing in most cases. If a @2x image isn't found, then Eclipse will scale up a normal image.

Your Mac isn't hidpi by default? I thought most macs were.

ilg-ul, 2020-11-29:
My Mac is, but the monitor is only 1920x1200.

jonahgraham, 2020-11-29:
There is some stuff [here](https://www.eclipse.org/eclipse/news/4.6/platform.php#swt-autoscale) - I think you can force scaling, but I haven't played with it much. 

This issue is done now. Thanks for merging the change.

closed 2020-11-29 by jonahgraham