eclipse-embed-cdt/eclipse-plugins #443: Invalid XML content for Eclipse Help

jonahgraham, 2020-11-27:
The `<?xml` tag is duplicated in the XML files for the eclipse help:

https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/d79986d43ec1d7e079eeeb58d68ebdb383a4ed2b/plugins%2Forg.eclipse.embedcdt.doc.user%2Ftoc.xml#L1-L5

https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/ceba9fd8419201f514d9a37b5e53266250ac6bdb/plugins%2Forg.eclipse.embedcdt.doc.user%2Ftopics_Concepts.xml#L1-L3

this causes errors when opening Eclipse help:
```
!ENTRY org.eclipse.help 4 0 2020-11-27 10:00:13.123
!MESSAGE Error reading help table of contents file /"org.eclipse.embedcdt.doc.user/toc.xml, URL = jar:file:/tmp/f/eclipse/plugins/org.eclipse.embedcdt.doc.user_2.0.0.202011221632.jar!/toc.xml at line 5, column 6" (skipping file)
!STACK 0
org.xml.sax.SAXParseException; lineNumber: 5; columnNumber: 6; The processing instruction target matching "[xX][mM][lL]" is not allowed.
	at java.xml/com.sun.org.apache.xerces.internal.parsers.DOMParser.parse(DOMParser.java:261)
	at java.xml/com.sun.org.apache.xerces.internal.jaxp.DocumentBuilderImpl.parse(DocumentBuilderImpl.java:339)
	at org.eclipse.help.internal.dynamic.DocumentReader.read(DocumentReader.java:59)
	at org.eclipse.help.internal.dynamic.DocumentReader.read(DocumentReader.java:47)
	at org.eclipse.help.internal.toc.TocFileParser.parse(TocFileParser.java:36)
	at org.eclipse.help.internal.toc.TocFileProvider.getTocContributions(TocFileProvider.java:49)
--snip--
```

ilg-ul, 2020-11-27:
Oh, thank you for catching this, I did not open the Eclipse Help for some time...

ilg-ul, 2020-11-28:
Fixed.

closed 2020-11-28 by ilg-ul