eclipse-embed-cdt/eclipse-plugins #444: ERROR: File .repos.xml does not exist, using defaults.

jonahgraham, 2020-11-27:
Sometimes there is an error in the error log that says:

```
org.eclipse.embedcdt.packs.data
 Error
 Fri Nov 27 04:50:30 CET 2020
 File .repos.xml does not exist, using defaults.
```

This is being reported as an error so shows as such in the .log or in the Error Log view. However it looks like a warning or info. Can the log call be changed?

https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/660f1c90e7f22d35674f6b33707ee452c8fbbfa8/plugins%2Forg.eclipse.embedcdt.packs.data%2Fsrc%2Forg%2Feclipse%2Fembedcdt%2Fpacks%2Fdata%2FRepos.java#L136-L139

https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/660f1c90e7f22d35674f6b33707ee452c8fbbfa8/plugins%2Forg.eclipse.embedcdt.core%2Fsrc%2Forg%2Feclipse%2Fembedcdt%2Fcore%2FAbstractActivator.java#L89-L91

jonahgraham, 2020-11-27:
I don't know if this is on topic or not, but the error message does not include a stack trace which makes it quite difficult to track down where in the code it came from.

ilg-ul, 2020-11-27:
Indeed, this looks like a warning. I'll fix it these days.

ilg-ul, 2020-11-28:
Fixed, the message was changed to Info after extending the log API via #446.

closed 2020-11-28 by ilg-ul