eclipse-embed-cdt/eclipse-plugins #445: Add files failure cannot read template source

jonahgraham, 2020-11-27:
Reproduced on 2020-12 M3 EPP on Linux:

If I create a new STM32F7xx C/C++ Project with default settings I get a template error at the end of the wizard. I am not 100% sure this is a Embed-CDT bug, it may be a CDT one in the template handler code?

It looks like there is a file case issue, the error says `timer-c.h` but the file is `Timer-c.h`:

https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/9c760c890cb9db1dd30c49e9548bd503c1bbd55b/plugins%2Forg.eclipse.embedcdt.templates.core%2Ftemplates%2Fcommon%2Finclude%2FTimer-c.h#L1

First page of wizard settings - the rest I choose defaults:

[image]

Error message:

[image]

Info from error log (seems to be the same as the message, except missing main bit about the timer):
```
!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.591
!MESSAGE Condition 1--> Process 1 (org.eclipse.cdt.managedbuilder.core.NewManagedProject)-->Success: Executed successfully with args: [name:hello, location:/tmp/f/data, artifactExtension:elf, isCProject:true]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.591
!MESSAGE Condition 2--> Process 1 (org.eclipse.embedcdt.templates.core.SetPropertyIfHasNature)-->Success: Executed successfully with args: [projectName:hello, natureString:org.eclipse.cdt.core.ccnature, propertyName:fileExtension, propertyValue:cpp]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.591
!MESSAGE Condition 2--> Process 2 (org.eclipse.cdt.core.AddNature)-->Success: Executed successfully with args: [projectName:hello, natureId:org.eclipse.cdt.core.ccnature]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.591
!MESSAGE Condition 2--> Process 3 (org.eclipse.embedcdt.templates.core.SetPropertyToPluginResource)-->Success: Executed successfully with args: [pluginId:org.eclipse.embedcdt.templates.core, relativePath:templates/common, propertyName:commonDir]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.591
!MESSAGE Condition 2--> Process 4 (org.eclipse.embedcdt.templates.core.ConditionalSetProperty)-->Success: Executed successfully with args: [projectName:hello, condition:, propertyName:cmsisFamilySymbol, propertyValue:STM32F745xx]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.597
!MESSAGE Condition 2--> Process 5 (org.eclipse.embedcdt.templates.core.ConditionalSetLowerCaseProperty)-->Success: Executed successfully with args: [projectName:hello, condition:, propertyName:device, propertyValue:STM32F745xx]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.598
!MESSAGE Condition 2--> Process 6 (org.eclipse.embedcdt.templates.core.ConditionalSetProperty)-->Success: Executed successfully with args: [projectName:hello, condition:, propertyName:content, propertyValue:blinky]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.599
!MESSAGE Condition 2--> Process 7 (org.eclipse.embedcdt.templates.core.ConditionalSetProperty)-->Success: Executed successfully with args: [projectName:hello, condition:, propertyName:hseValue, propertyValue:8000000]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.599
!MESSAGE Condition 3--> Process 1 (org.eclipse.cdt.core.CreateFolder)-->Success: Executed successfully with args: [projectName:hello, path:include]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.599
!MESSAGE Condition 3--> Process 2 (org.eclipse.cdt.managedbuilder.core.AppendToMBSStringListOptionValues)-->Success: Executed successfully with args: [projectName:hello, resourcePaths:{{id:ilg.gnuarmeclipse.managedbuild.cross.option.*.include.paths, values:{"../include"}, path:}}]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.599
!MESSAGE Condition 4--> Process 1 (org.eclipse.cdt.core.CreateFolder)-->Success: Executed successfully with args: [projectName:hello, path:system/include/cmsis]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.600
!MESSAGE Condition 4--> Process 2 (org.eclipse.cdt.managedbuilder.core.AppendToMBSStringListOptionValues)-->Success: Executed successfully with args: [projectName:hello, resourcePaths:{{id:ilg.gnuarmeclipse.managedbuild.cross.option.*.include.paths, values:{"../system/include"}, path:}, {id:ilg.gnuarmeclipse.managedbuild.cross.option.*.include.paths, values:{"../system/include/cmsis"}, path:}}]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.600
!MESSAGE Condition 5--> Process 1 (org.eclipse.cdt.core.CreateFolder)-->Success: Executed successfully with args: [projectName:hello, path:system/include/stm32f7-hal]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.600
!MESSAGE Condition 5--> Process 2 (org.eclipse.cdt.managedbuilder.core.AppendToMBSStringListOptionValues)-->Success: Executed successfully with args: [projectName:hello, resourcePaths:{{id:ilg.gnuarmeclipse.managedbuild.cross.option.*.include.paths, values:{"../system/include/stm32f7-hal"}, path:}}]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.600
!MESSAGE Condition 6--> Process 1 (org.eclipse.cdt.core.CreateSourceFolder)-->Success: Executed successfully with args: [projectName:hello, path:src]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.600
!MESSAGE Condition 7--> Process 1 (org.eclipse.cdt.core.CreateSourceFolder)-->Success: Executed successfully with args: [projectName:hello, path:system]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.601
!MESSAGE Condition 8--> Process 1 (org.eclipse.embedcdt.templates.core.ConditionalAddFiles)-->Success: Executed successfully with args: [projectName:hello, condition:, files:{{source:/tmp/f/eclipse/plugins/org.eclipse.embedcdt.templates.core_3.0.0.202011221632/templates/common//src/main/main-100-include.c, target:src/main.c, replaceable:true}}]

!ENTRY org.eclipse.cdt.core 1 0 2020-11-27 10:28:17.601
!MESSAGE Condition 8--> Process 2 (org.eclipse.embedcdt.templates.core.ConditionalAppendCreate)-->Success: Executed successfully with args: [projectName:hello, condition:, files:{{source:/tmp/f/eclipse/plugins/org.eclipse.embedcdt.templates.core_3.0.0.202011221632/templates/common//src/main/main-110-include-blinky.c, target:src/main.c, replaceable:true}, {source:/tmp/f/eclipse/plugins/org.eclipse.embedcdt.templates.core_3.0.0.202011221632/templates/common//src/main/main-120-none-blinky.c, target:src/main.c, replaceable:true}, {source:/tmp/f/eclipse/plugins/org.eclipse.embedcdt.templates.core_3.0.0.202011221632/templates/common//src/main/main-130-trace-via.c, target:src/main.c, replaceable:true}, {source:/tmp/f/eclipse/plugins/org.eclipse.embedcdt.templates.core_3.0.0.202011221632/templates/common//src/main/main-140-defs-blinky.c, target:src/main.c, replaceable:true}, {source:src/main/main-145-leds-blinky.c, target:src/main.c, replaceable:true}, {source:/tmp/f/eclipse/plugins/org.eclipse.embedcdt.templates.core_3.0.0.202011221632/templates/common//src/main/main-150-begin.c, target:src/main.c, replaceable:true}, {source:/tmp/f/eclipse/plugins/org.eclipse.embedcdt.templates.core_3.0.0.202011221632/templates/common//src/main/main-180-none-hello.c, target:src/main.c, replaceable:true}, {source:/tmp/f/eclipse/plugins/org.eclipse.embedcdt.templates.core_3.0.0.202011221632/templates/common//src/main/main-190-clock-blinky.c, target:src/main.c, replaceable:true}, {source:src/main/main-200-none-loop-blinky.c, target:src/main.c, replaceable:true}, {source:/tmp/f/eclipse/plugins/org.eclipse.embedcdt.templates.core_3.0.0.202011221632/templates/common//src/main/main-230-none-end.c, target:src/main.c, replaceable:true}}]

```

ilg-ul, 2020-11-27:
It is a template error. This bug was introduced when I reworked the templates.

ilg-ul, 2020-11-28:
Fixed.

The problem was caused by a failed rename of the file.

For historical reasons, the macOS filesystem is in fact case insensitive, although the case is preserved and shown properly; for unknown reasons, Git checks the case insensitive names, and simply renaming a file with a different case does not make it dirty, so it is not committed.

closed 2020-11-28 by ilg-ul