eclipse-embed-cdt/eclipse-plugins #457: Reformat code to CDT style

ilg-ul, 2020-12-06:
At a certain moment the CDT code was reformatted to a given style.

Can we do the same for the Embed CDT code?

jonahgraham, 2020-12-06:
Yes - I can do that. CDT is very strict now and will reject a PR that breaks the code formatting and whitespace. We can adapt that here is desired too.

One change means that the .settings for each project should be checked in and the project will store the formatting settings.

Big problem is I don't know if I can do this before 6.0.0 (Tuesday). I will try though and give an update later.

ilg-ul, 2020-12-06:
Please merge everything to develop and create a new branch from there.

ilg-ul, 2020-12-06:
> CDT is very strict now and will reject a PR that breaks the code formatting and whitespace. We can adapt that here is desired too

Yes, that would be nice, but I guess we can do it in a second step, for now simply reformatting everything would be enough.

ilg-ul, 2020-12-06:
At a second thought I guess we can do this after Tuesday, better to keep the code ready in case we need to patch something else before Tuesday.

ilg-ul, 2020-12-06:
My simplistic view of the subject is that we take the `formatsettings.xml` from CDT, select everything and with Source -> Format fix all files.

If this is not realistic, we'll definitely leave it for after Tuesday.

ilg-ul, 2020-12-29:
Jonah, I think we're ready with all changes, and we can proceed with code reformatting.

jonahgraham, 2020-12-30:
> My simplistic view of the subject was that we take the `formatsettings.xml` from CDT, select everything and with Source -> Format fix all files.

That is one option - I provided in #469  what we do for CDT which is more heavy weight, but helps to keep the code formatted by other contributors in sync with embedcdt because the settings are stored with projects instead of the workspace.

commit 7115cc99 (2021-01-06): Merge pull request #469 from eclipse-embed-cdt/jonah/bug_457_code_cleanup (fixes #457)

Code formatting and cleanup - fixes Bug #457
closed 2021-01-06 by jonahgraham