eclipse-embed-cdt/eclipse-plugins #459: Add API Baseline setup to build prerequistes

jonahgraham, 2020-12-07:
A lot of work has been done in 6.0.0 to have API better defined. However to make this successful, embedcdt then has to verify that there is an API baseline defined in the workspace that Eclipse API tools can compare against.

At the moment on https://eclipse-embed-cdt.github.io/develop/build-prerequisites/#install-base-api-cdt-1000 only CDT is in the baseline, so that means none of the embedcdt plug-ins will be checked for API compatibility issues. Once 6.0.0 is released, the API baseline needs to point at the embedcdt release.

This is best done with a embedcdt-baseline.target file that points at the p2 sites of the baseline.

ilg-ul, 2020-12-07:
Well, this sounds nice, but what exactly does it mean? 

Who creates this file, where should it be located, who must use it and how? (lots of questions)

jonahgraham, 2020-12-07:
"we" create the file - it should be in git.

If the API baseline is properly configured then it means that extenders of embedcdt can rely on the version number to follow semantic versioning because we as developers will get an error if we introduce a change that breaks semantic versioning.

BTW I think there should have been a warning/error up until now related to this from Eclipse. I raised https://bugs.eclipse.org/bugs/show_bug.cgi?id=569528 for this issue.

ilg-ul, 2020-12-07:
In my development Eclipse I see three errors:

```
Description	Resource	Path	Location	Type
Plugin execution not covered by lifecycle configuration: org.apache.maven.plugins:maven-clean-plugin:2.5:clean (execution: default-clean-1, phase: initialize)	pom.xml	/org.eclipse.embedcdt-repository	line 23	Maven Project Build Lifecycle Mapping Problem
Plugin execution not covered by lifecycle configuration: org.eclipse.tycho:tycho-packaging-plugin:1.7.0:build-qualifier-aggregator (execution: default-build-qualifier-aggregator, phase: validate)	pom.xml	/org.eclipse.embedcdt-repository	line 25	Maven Project Build Lifecycle Mapping Problem
There is a possible API baseline mismatch since none of the workspace projects is in the baseline.			Unknown	Default API Baseline Problem
```

Perhaps the last one is what you would expect?

ilg-ul, 2020-12-07:
BTW, I welcome these improvements, but, when ready, we should also mention these details in the documentation.

jonahgraham, 2020-12-07:
> Perhaps the last one is what you would expect?

Yes - that is the error. But I don't get that error - details in Bug 569528 

The Plugin execution not covered by lifecycle configuration is why I don't use m2e (see #458 item 4).

jonahgraham, 2020-12-07:
> BTW, I welcome these improvements, but, when ready, we should also mention these details in the documentation.

Of course!

ilg-ul, 2020-12-07:
Please add a ticket to the web-jekyll repo to remember this.

PR #475 (2021-01-08): Add API Baseline support
This fixes #459 

See the individual commit comments for more details.
closed 2021-01-08 by ilg-ul