eclipse-embed-cdt/eclipse-plugins #461: On the problem of not displaying the address bar icon in Linux

laomaolaile, 2020-12-10:
When I was using Eclipse IDE for Embedded C/C++ Developers 2020-09 in Linux, I found that the address bar icon could not be found.What can I do to make that icon display？

ilg-ul, 2020-12-10:
There were some major changes to the project, could you install a separate Eclipse with the latest 2020-12 RC1 from 

- https://ci.eclipse.org/packaging/job/simrel.epp-tycho-build/lastSuccessfulBuild/artifact/org.eclipse.epp.packages/archive/

and post some screen capture with the problem?

Please be sure you also create a new workspace and import your project, otherwise you might need to reset the perspectives to display properly.

laomaolaile, 2020-12-10:
The package I downloaded is
[image]

The address bar icon is shown in the figure below， could not be found.
[image]

ilg-ul, 2020-12-10:
I'm afraid this is an Ubuntu setting which will require the help of an Ubuntu specialist, which I'm not. :-(

How did you create that entry in the tool bar? Perhaps there is a way to also set the icon.

TommyMurphyTM1234, 2020-12-10:
You probably want to create a desktop icon file for the application and then drag it to the dock bar.

https://www.howtogeek.com/445303/how-to-create-desktop-shortcuts-on-ubuntu/

As Liviu says this is not really a plugin issue.

ilg-ul, 2020-12-11:
I don't know if this applies to Ubuntu, but I just tested the Eclipse Installer on Raspberry Pi OS, and it created an executable file on the desktop with the following content:

```
#!/usr/bin/env xdg-open
[Desktop Entry]
Type=Application
Terminal=false
Encoding=UTF-8
Version=1.1
Name=Eclipse IDE for Embedded C/C++ Developers - 2020-12
Exec=env GTK_IM_MODULE=ibus /home/ilg/eclipse/embedcpp-2020-12/eclipse/eclipse
Categories=Development;IDE;
Icon=/home/ilg/eclipse/embedcpp-2020-12/eclipse/icon.xpm
```

closed 2020-12-20 by ilg-ul