eclipse-embed-cdt/eclipse-plugins #462: A problem about openocd

laomaolaile, 2020-12-15:
[image]

When I created a project, I reported an error, as shown in the figure. This error does not appear every time. It is random and can be encountered occasionally. How can this be solved.

laomaolaile, 2020-12-15:
[image]

ilg-ul, 2020-12-15:
What template project are you using?

laomaolaile, 2020-12-16:
It's a new template I created myself.
Under what circumstances will this error be corrected?

ilg-ul, 2020-12-16:
Starting with the (just released) v6.0.0, the project was greatly restructured, so I suggest you first update your template to the new Java package names and Eclipse IDs.

closed 2020-12-20 by ilg-ul