eclipse-embed-cdt/eclipse-plugins #468: Multithreaded CMSIS-Pack index update

flit, 2020-12-29:
The CMSIS-Pack plugin should use multithreaded downloads of PDSC files to update the index.

For comparison, [cmsis-pack-manager](https://github.com/pyocd/cmsis-pack-manager/) can update the entire index (currently 695 packs) in less than 20 seconds, including download and parsing the PDSC. Most of that time is waiting for a request for an invalid pack URL to time out; the actual update time for 694 packs is about 3 seconds.

ilg-ul, 2020-12-30:
Yes, that would be interesting to know how they did it.

Feel free to investigate and suggest solutions.

flit, 2020-12-30:
If I get a chance, I'll try to work on it.

ilg-ul, 2020-12-30:
I suggest you first make a plan, and then we discuss possible solutions.

flit, 2020-12-30:
Sounds good.

ilg-ul, 2020-12-30:
Don't get too excited, I completely forgot how that code works, so I can't be of much practical help. :-(

Plus that, as I mentioned, as soon as I reorganise my projects, I'll no longer use CMSIS Packs at all, so my interest in spending time on them is pretty low.

flit, 2020-12-30:
Understood. My interest is that users can take advantage of CMSIS Packs to get both SVDs and pyocd config data via an integrated UI. I don't need it for myself (cmsis-pack-manager is integrated into pyocd). (I'm also thinking about making the SVD from pyocd available to the launch config.)

ilg-ul, 2020-12-30:
The J-Link plug-in automatically accesses the SVD if the device name was correctly associated with the project and the CMSIS Pack is available.

Anyway, an alternate solution with only a bit more effort to configure it is to copy the SVD in the project, and configure the path to it in the SVD tab. And since 6.1.0, the SVD can also be compressed with zip.

flit, 2020-12-30:
Nice, thanks for the tips.