eclipse-embed-cdt/eclipse-plugins #477: Embed CDT description missing pyOCD plugin

flit, 2021-01-08:
The [description of the Embedded CDT package on eclipse.org](https://www.eclipse.org/downloads/packages/release/2020-12/r/eclipse-ide-embedded-cc-developers) is missing the pyOCD plugin. This description also appears in the Eclipse Installer.

The current text is: "It includes managed cross build plug-ins (Arm and RISC-V) and debug plug-ins (SEGGER J-Link, OpenOCD, and QEMU), plus a number of templates to create ready to run blinky projects. "

This should be updated to: "…debug plug-ins (SEGGER J-Link, OpenOCD, _pyOCD_, and QEMU)…"

ilg-ul, 2021-01-08:
Fixed: https://git.eclipse.org/r/c/epp/org.eclipse.epp.packages/+/174506

However it'll be effective starting with 2021-03.

closed 2021-01-08 by ilg-ul

flit, 2021-01-08:
Thanks a lot!