eclipse-embed-cdt/eclipse-plugins #483: Feature request: add input field for choosing device in debug configuration

jope12, 2021-01-19:
Hello,

I have a feature request for the "Debug Configurations".
Although it is possible to use Eclipse as a GUI frontend for GDB without creating a project, when also using a SVD file, creating a project is obligatory. One can associate a SVD file with a debug configuration, but it is also needed to create a project file (which can be empty) and then to choose a device, otherwise the "Peripherals Register" view will stay empty.

My request would be to add an input field to the debug configuration to be able to choose a device without needing to explicitly create a project (but in addition to a possible selection by a project), to make it possible to use the SVD file and therefore the "Peripherals Register" view.

ilg-ul, 2021-01-19:
Thank you for your request, I'll consider it for a future release.

flit, 2021-01-26:
I need to double check, but I'm pretty sure I'm successfully using the Peripheral Registers view with a pyocd launch config where the SVD is set in the launch config and the project has no device set. This is an external make project though, not managed make.

ilg-ul, 2021-01-26:
I did not check the SVD content recently, but, if I remember right, SVD files may define registers for multiple devices.

In this case there must be a way to select which is the desired device, otherwise the plug-in will most probably use a default, which might not be what you want.

flit, 2021-01-26:
> SVD files may define peripherals for multiple devices.

Nope. (Thankfully 😄)

From the [SVD docs](https://arm-software.github.io/CMSIS_5/SVD/html/elem_device.html): "Only one <device> section is allowed per file."

ilg-ul, 2021-01-26:
Great!

Then there is no need for an additional field in the interface.

And we can close this issue.

flit, 2021-01-26:
I still need to double check that I hadn't selected a device without remembering it!

flit, 2021-01-26:
Just checked, I don't have a device selected in the projects I was referencing. Again, this is with the external builder.

jope12, 2021-01-27:
>Then there is no need for an additional field in the interface.
>And we can close this issue.

Does this mean that this will not be implemented generally or only not for pyocd?

Because when I choose "GDB OpenOCD Debugging", and then choose a SVD file, my peripherals view still stays empty.
I'm forced to create an empty project and choose a device.

ilg-ul, 2021-01-27:
It means that it looks like there is no need to add a new input field, manually selecting the SVD file should be enough.

The reason why an empty project is still needed is another issue, to be investigated.

jope12, 2021-01-27:
Ok, thanks. This would be even better then.

danielRep, 2023-10-04:
This would be an awesome feature!