eclipse-embed-cdt/eclipse-plugins #484: Embed CDT 6.0.0 does not upgrade to 6.1.0

jonahgraham, 2021-01-20:
Try installing Embed CDT 6.0.0 from SimRel into [Eclipse Platform](https://www.eclipse.org/downloads/download.php?file=/eclipse/downloads/drops4/R-4.18-202012021800/eclipse-platform-4.18-linux-gtk-x86_64.tar.gz) and then after restart do a check for updates and instead of a clean update to 6.1.0 (whose p2 site is automatically added by installing 6.0.0)

This is what user is presented with

[image]

This is the underlying error that leads to the above bad p2 solution

```
!ENTRY org.eclipse.equinox.p2.operations 4 0 2021-01-19 19:35:33.265
!MESSAGE Operation details
!SUBENTRY 1 org.eclipse.equinox.p2.director 4 10054 2021-01-19 19:35:33.265
!MESSAGE Cannot complete the install because of a conflicting dependency.
!SUBENTRY 2 org.eclipse.equinox.p2.director 4 0 2021-01-19 19:35:33.265
!MESSAGE Software being installed: Embedded C/C++ STM32Fx Project Templates 6.1.0.202101081915 (org.eclipse.embedcdt.templates.stm.feature.group 6.1.0.202101081915)
!SUBENTRY 2 org.eclipse.equinox.p2.director 4 0 2021-01-19 19:35:33.265
!MESSAGE Software currently installed: Embedded C/C++ Core 6.0.0.202012080907 (org.eclipse.embedcdt.feature.group 6.0.0.202012080907)
!SUBENTRY 2 org.eclipse.equinox.p2.director 4 1 2021-01-19 19:35:33.265
!MESSAGE Only one of the following can be installed at once: 
!SUBENTRY 3 org.eclipse.equinox.p2.director 4 0 2021-01-19 19:35:33.265
!MESSAGE Embedded C/C++ Core Plug-in 6.0.0.202012080907 (org.eclipse.embedcdt.core 6.0.0.202012080907)
!SUBENTRY 3 org.eclipse.equinox.p2.director 4 0 2021-01-19 19:35:33.265
!MESSAGE Embedded C/C++ Core Plug-in 6.1.0.202101081915 (org.eclipse.embedcdt.core 6.1.0.202101081915)
!SUBENTRY 2 org.eclipse.equinox.p2.director 4 1 2021-01-19 19:35:33.265
!MESSAGE Cannot satisfy dependency:
!SUBENTRY 3 org.eclipse.equinox.p2.director 4 0 2021-01-19 19:35:33.265
!MESSAGE From: Embedded C/C++ Core 6.0.0.202012080907 (org.eclipse.embedcdt.feature.group 6.0.0.202012080907)
!SUBENTRY 3 org.eclipse.equinox.p2.director 4 0 2021-01-19 19:35:33.265
!MESSAGE To: org.eclipse.equinox.p2.iu; org.eclipse.embedcdt.core [6.0.0.202012080907,6.0.0.202012080907]
!SUBENTRY 2 org.eclipse.equinox.p2.director 4 1 2021-01-19 19:35:33.265
!MESSAGE Cannot satisfy dependency:
!SUBENTRY 3 org.eclipse.equinox.p2.director 4 0 2021-01-19 19:35:33.265
!MESSAGE From: Embedded C/C++ STM32Fx Project Templates 6.1.0.202101081915 (org.eclipse.embedcdt.templates.stm.feature.group 6.1.0.202101081915)
!SUBENTRY 3 org.eclipse.equinox.p2.director 4 0 2021-01-19 19:35:33.265
!MESSAGE To: org.eclipse.equinox.p2.iu; org.eclipse.embedcdt.core [6.1.0.202101081915,6.1.0.202101081915]
```

jonahgraham, 2021-01-20:
I reproduced the error using  Eclipse Platform to make sure that there was the minimum number of other things that could be interfering. The same (or at least very similar?) thing happens if Embedded C/C++ EPP package is used and check for updates is run.

jonahgraham, 2021-01-20:
If I install only the *Embedded C/C++ Core* feature then the Check for Updates says there is nothing to install. But if I go to Install new software I can selected Embedded C/C++ Core feature and it is converted to an upgrade.

[image]

jonahgraham, 2021-01-20:
@ilg-ul I found the problem. The issue is a bug in the p2.inf to handle the name change

The org.eclipse.embedcdt feature when it was created I assume was copied from the codered feature in #426 . That meant the p2.inf instruction in it is incorrect as it says it can update codered:

https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/4982fd8e7b129e9c8c33ba30f1f9fb5392abc4f3/features/org.eclipse.embedcdt-feature/p2.inf#L10

Since this feature didn't exist before 6.0.0, the p2.inf's update line quoted above can just be removed.

jonahgraham, 2021-01-20:
I fixed this in 7ff8bf1260b0bdcbbf7c3e535ee64779fd07d27c - with this patch the updates work as expected. However if you also have the published 6.1.0 update site available it will initially update fine, but then you get this error on check for updates:

[image]

Now we have to decide how to resolve the problem going forward for users. I recommend doing a 6.1.1 or 6.2.0 release and remove the 6.1 from the composite site.

ilg-ul, 2021-01-20:
Thank you Jonah, I'll make a new 6.1.1 and remove 6.1.0 from the composite site.

ilg-ul, 2021-01-20:
Fixed on 2020-01-20.

closed 2021-01-20 by ilg-ul