eclipse-embed-cdt/eclipse-plugins #487: Bug: CMSIS packs menu entry is shown in all perspectives

marcel-kanter, 2021-02-22:
The Embedded Tools install the CMSIS packs package. If you install the embedded tools into an eclipse with other perspectives, the "Make CMSIS packs perspective" tool bar is visible in ALL perspectives AND the main all-perspective-closed view of eclipse.

This is a bit anoying to have a toolbar floating around that should simply show an perspective, especially when it does not work from the all closed view. Please remove this toolbar.

Additionally the name in the toolbar configuration is not proper and shows the internal name.

ilg-ul, 2021-02-22:
Can you provide some images and some more details?

marcel-kanter, 2021-02-25:
- Download http://download.eclipse.org/eclipse/downloads/drops4/R-4.18-202012021800/download.php?dropFile=eclipse-platform-4.18-win32-x86_64.zip
- Extract to some directory
- Start eclipse/eclipse.exe
- Chose a directory for a fresh workspace & click launch
--> Eclipse opens with welcome screen
- Click close on welcome screen
--> Eclipse shows "Resource" Perspective
- Click on "Help" -> "Install new Software"
- Choose in the drop-down "Work with:" the entry "2020-12 - http://download.eclipse.org/releases/2020-12" or "--all available sites--"
- Enter "Embedded" into search bar below and hit enter
- Choose only "Embedded C/C++ Arm Cross Compiler" and "Embedded C/C++ Core" Leave "Embedded C/C++ CMSIS Packs (Experimental) " unselected

[image]

- Verify "2 items selected" right below the selecting box
- Click "Next", Click "Next", Choose "Accept the terms of license aggreement" and click "Finish"
- Wait for the packets to get downloaded and installed (process bar is in the bottom right by default)
- Click "Restart Now" when it asks for the restart
- Wait for eclipse to re-open

- See the "Make CMSIS Packs perspective visible toolbar" in Resource Perspective

[image]

- Click on The Button and the perspective gets opended besides the fact that it was not selected and is officially not installed as package

- Verify this by clicking on "Help" -> "About Eclipse Platform", click on "Installation Details"

[image]

- Click on close of the opened dialogs and close the editor or what this is in the cmsis packs perspective

[image]

- Click on "Window" -> "Perspective" -> "Close All Perspectives"

The Toolbar is still there

[image]
[image]

- Click on the Icon "Make CMSIS Packs Perspective visible" and nothing happens.

(I dont know how to make a screen shot of something that does not show up)

To summarize:

**The jar files** 

	Eclipse Embedded CDT	Embedded C/C++ CMSIS Packs Core Plug-in	3.0.0.202012080907	org.eclipse.embedcdt.packs.core
	Eclipse Embedded CDT	Embedded C/C++ CMSIS Packs UI Plug-in	2.0.0.202012080907	org.eclipse.embedcdt.packs.ui

are installed even when CMSIS packs are not selected for install. You can find the versions in "Help" -> "About Eclipse Platform" -> "Installation Details" -> "Plug-Ins". Or by extraction of the jar files in the "plugins" subfolder where you have extracted the zip file with the platform.

This tab of the this dialog basically shows all versions of all jar files in the plugins/dropins folder.

**The toolbar isnt working when all perspectives are closed**

This may be inteded or a bug.

**The toolbar is added all perspectives"**

To verify this:
- Click on "Window" -> "Perspective" -> "Open Perspective" -> "Other"
- Choose "Team synchronizing"

The toolbar is there
[image]

- Repeat the steps for "Resource (default)" perspective or any other perspective already created.

[image]

**The localization for the Toolbar seems to be missing or the description is missing**

- Click on "Window" -> "Perspective" -> "Customize Perspective"

[image]

ilg-ul, 2021-02-25:
Thank you, for the detailed bug report, it should be enough to reproduce the problem.

At first sight there might be multiple small issues, like unwanted dependencies, or missing conditions.

I'll investigate.

ilg-ul, 2021-03-01:
I more or less reproduced the problem.

However, in two tests, one on Windows and one on macOS, the initial view of the CMSIS Packs perspective was correct, it showed an empty window, instructing to press the Refresh button. I guess you already had an `AppData\Roaming\CMSIS-Packs` folder, possibly with inconsistent content. I suggest you rename/remove the `CMSIS Packs` folder, and reinstall the Eclipse + plug-ins from scratch.

Putting this aside, I summarise your report as several separate issues:

- the Windows -> Perspectives -> Close All Perspectives leaves the Packs button enabled in the toolbar, and this button is not functional, it does not reopen the perspective; to reopen it is necessary to do  Windows -> Perspectives -> Open Perspective
- the Packs perspective should not be visible if the **Embedded C/C++ CMSIS Packs (Experimental)** feature is not installed
- the perspective description is missing and replaced by _showPerspective_

Can you confirm this?

ilg-ul, 2021-03-01:
The missing label was fixed as the separate #488.

New logic was added to the toolbar button to make it visible only for C/C++ projects (having the `org.eclipse.cdt.core.cnature` nature).

Unfortunately the current code in the managed build (Arm & RISC-V) UI plug-ins have a dependency on the packs UI plug-in, which also display the CMSIS Packs perspective, and hiding this perspective completely requires moving the perspective code to a separate plug-in.

ilg-ul, 2021-03-01:
A preview of the next 6.1.3 is available for testing from:

- https://download.eclipse.org/embed-cdt/builds/develop/p2/

Please install it and let me know is this works for you.

marcel-kanter, 2021-03-01:
> I more or less reproduced the problem.
>
> However, in two tests, one on Windows and one on macOS, the initial view of the CMSIS Packs perspective was correct, it showed an empty window, instructing to [...]

Yes. As you described. BTW I do not have any packs installed. So the packs view maybe empty correctly.

(I'm a bit busy these days to excuse my late replies)

ilg-ul, 2021-03-01:
No problem, the `develop/p2` site will remain available, so please test it when possible.

marcel-kanter, 2021-03-20:
So far so good. It seems to work now. Name is shown correctly and it is not appearing on all perspectives when no CMSIS packs packet is installed. The CMSIS Packs perspective still can be opened (as per #489)

The logic with the project (action sets) seems to work - even when switching between general purpose project type and c/c++ project. But the customize dialog shows a bit strange Tool-Tip: (Cosmetics)

[image]

ilg-ul, 2021-03-20:
Ok, thank you for confirming.

ilg-ul, 2021-03-20:
Fixed on 2021-03-01.

closed 2021-03-20 by ilg-ul