eclipse-embed-cdt/eclipse-plugins #491: The peripheral view is empty

zhaoxuji, 2021-06-07:
The problem description is like the title.
[image]
[image]

[log.txt](https://github.com/eclipse-embed-cdt/eclipse-plugins/files/6605422/log.txt)

ilg-ul, 2021-06-07:
Your SVD files seem to contain non-parseable characters.

jonahgraham, 2021-06-15:
@zhaoxuji Are you able to share the SVD file you used? The Xml reader (linked below) should be handling the encoding as declared in the SVD XML file. Is there a `encoding="utf-8"` or similar in the SVD? And if so, does it match the contents of the file?

https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/c57bac116b482ee102ebbeb875efdaea1cf25b32/plugins/org.eclipse.embedcdt.core/src/org/eclipse/embedcdt/core/Xml.java#L50-L59

SoniaZotz, 2022-01-10:
having the same problem
[image]
 
[image]

TommyMurphyTM1234, 2022-01-11:
> having the same problem

You probably need to post your SVD XML file.

SoniaZotz, 2022-01-11:
standard ARM  Pscks -- Arm\Packs\NordicSemiconductor\nRF_DeviceFamilyPack\8.44.1\SVD
[nrf51.zip](https://github.com/eclipse-embed-cdt/eclipse-plugins/files/7850561/nrf51.zip)

ilg-ul, 2022-01-12:
The file is ok, it was parsed properly:

[image]

SoniaZotz, 2022-01-13:
yes thank -its working  I used the wrong file in the SVD  selection

closed 2022-04-01 by ilg-ul