eclipse-embed-cdt/eclipse-plugins #492: Not able to pass the commands to gdb client (In openocd plugin)

vidyatc, 2021-06-17:
When we pass commands in Commands(text field) and other options (text field) in the Debugger tab, the plugin does not accept those commands.

[image]

* plug-in version  - openocd 5.2.1
* Eclipse version - 2020-09 (4.17.0)
* operating system - Ubuntu 18

ilg-ul, 2021-06-17:
> the plugin does not accept those commands.

Can you be more specific and describe what happens?

vidyatc, 2021-06-17:
if we uncheck Load symbol and Load executable in startup and pass "load < file.exe >"  in commands(text field) it's throwing Failed to execute MI command
Error in final launch sequence
Failed to execute MI command:
load /home/performance-server/openocd/freedom-e-sdk/software/hello/hello
Error message from debugger back end:
You can't do that when your target is `None'
Failed to execute MI command:
load /home/performance-server/openocd/freedom-e-sdk/software/hello/hello
Error message from debugger back end:
You can't do that when your target is `None'
You can't do that when your target is `None'

ilg-ul, 2021-06-17:
Can you compare the GDB trace output between the two cases?

What openocd binary are you using?

TommyMurphyTM1234, 2021-06-17:
I suspect that putting the load command in the Debugger > GDB Client Setup > Commands field means that it happens before the target remote connection has been established. Have you tried putting it in the Startup > Initialization Commands section instead?

TommyMurphyTM1234, 2021-06-18:
@vidyatc - did you try that I described?
@ilg-ul - I am pretty sure that this is not an issue with the Eclipse Embedded CDT plugins. In fact not an issue at all. :-)

jonahgraham, 2021-06-18:
Thanks @TommyMurphyTM1234.

@vidyatc if there is more info that indicates a problem in embed cdt we can reopen. Let us know either way as it can help the next person who has a similar error.

closed 2021-06-18 by jonahgraham