eclipse-embed-cdt/eclipse-plugins #530: Resolve reading Write-Only registers in peripheral register view

promodkumar-ashling, 2022-06-27:
### Description

The write-only registers(The access type information available in the peripheral register file) attempt to read values from UI when the peripheral register views are opened.

### Steps to Reproduce
1. Run an Arm GDB Launch.
2. Enable the peripheral register view.
3. Keep a register with "access" as "write-only" in the SVD Register file.
Eg: 
[image]
5. Try to open the register with access write-only.

**Expected behavior:** [What you expected to happen]
It shouldn't read the "write-only" register values.

**Actual behavior:** [What actually happened]
The write-only registers attempt to read values from UI causing the target to go for an exception.

### Versions

* EmbedCDT - 6.2.1
* Eclipse - 2022-03
* Java - 16
* Windows

promodkumar-ashling, 2022-08-10:
[image]

promodkumar-ashling, 2022-08-10:
```
                    <register>
   				<name>SMMU_SGFSR</name>
   				<displayName>Secure Global Fault Status register</displayName>
   				<description>Gives the fault status for each of the following possible faults.</description>
   				<addressOffset>0x48</addressOffset>
   				<size>32</size>
   				<access>write-only</access>
   				<resetValue>0x00000000</resetValue>
   				<resetMask>0xffffffff</resetMask>	
   				<fields>
   					<field>
   						<name>MULTI</name>
   						<lsb>31</lsb>
   						<msb>31</msb>
   						<access>write-only</access>
   						<modifiedWriteValues>modify</modifiedWriteValues>
   					</field>
   					<field>
   						<name>UUT</name>
   						<lsb>8</lsb>
   						<msb>8</msb>
   						<access>write-only</access>
   						<modifiedWriteValues>modify</modifiedWriteValues>
   					</field>
   					<field>
   						<name>PF</name>
   						<lsb>7</lsb>
   						<msb>7</msb>
   						<access>write-only</access>
   						<modifiedWriteValues>modify</modifiedWriteValues>
   					</field>
   					<field>
   						<name>EF</name>
   						<lsb>6</lsb>
   						<msb>6</msb>
   						<access>write-only</access>
   						<modifiedWriteValues>modify</modifiedWriteValues>
   					</field>
   					<field>
   						<name>CAF</name>
   						<lsb>5</lsb>
   						<msb>5</msb>
   						<access>write-only</access>
   						<modifiedWriteValues>modify</modifiedWriteValues>
   					</field>
   					<field>
   						<name>UCIF</name>
   						<lsb>4</lsb>
   						<msb>4</msb>
   						<access>write-only</access>
   						<modifiedWriteValues>modify</modifiedWriteValues>
   					</field>
   					<field>
   						<name>UCBF</name>
   						<lsb>3</lsb>
   						<msb>3</msb>
   						<access>write-only</access>
   						<modifiedWriteValues>modify</modifiedWriteValues>
   					</field>
   					<field>
   						<name>SMCF</name>
   						<lsb>2</lsb>
   						<msb>2</msb>
   						<access>write-only</access>
   						<modifiedWriteValues>modify</modifiedWriteValues>
   					</field>
   					<field>
   						<name>USF</name>
   						<lsb>1</lsb>
   						<msb>1</msb>
   						<access>write-only</access>
   						<modifiedWriteValues>modify</modifiedWriteValues>
   					</field>
   					<field>
   						<name>ICF</name>
   						<lsb>0</lsb>
   						<msb>0</msb>
   						<access>write-only</access>
   						<modifiedWriteValues>modify</modifiedWriteValues>
   					</field>
   				</fields>
   			</register>
```

promodkumar-ashling, 2024-10-15:
@ilg-ul Any updates on this issue?

ilg-ul, 2024-10-15:
No, but, as Jonah mentioned, we'll be happy to accept patches.