eclipse-embed-cdt/eclipse-plugins #540: Setting breakpoint at Reset_Handler stalls in fpu init

XaviLC, 2022-09-06:
### Description
Setting a break point on Reset_Handler seems to stall at fpu init and not going to main using Jlink with cortex m7, however if I specify the program counter at the right address of the main , then I see the starting up going through. To illustrate this,  I am setting a break point in my reset handler, as shown in below picture
[image]

The reset handler is defined as per

```C
void Reset_Handler( void )
{

    rfeMPU_init();

    rfeSwStartup_enableFPU();

    main( 0, 0 );
}
```
What I observe is that the debugger seems to stall in the fpu init, hitting the resume button doesn't help and same debug message is printed in the Console ( beneath correspond to two clicks on resume button)
[image]
```
Setting breakpoint @ address 0x000008D2, Size = 2, BPHandle = 0x002D
Starting target CPU...
...Target halted (DBGRQ, PC = 0x00011F9C)
Reading all registers
Removing breakpoint @ address 0x000008D2, Size = 2
Read 4 bytes @ address 0x00011F9C (Data = 0x0A10EEF1)
Read 4 bytes @ address 0x000000A2 (Data = 0x49034803)
Reading 64 bytes @ address 0x00000080
Read 4 bytes @ address 0x000000A2 (Data = 0x49034803)
Read 4 bytes @ address 0x20000534 (Data = 0x00000400)
Setting breakpoint @ address 0x000008D2, Size = 2, BPHandle = 0x002E
Starting target CPU...
...Target halted (DBGRQ, PC = 0x00011F9C)
Reading all registers
Removing breakpoint @ address 0x000008D2, Size = 2
Read 4 bytes @ address 0x00011F9C (Data = 0x0A10EEF1)
Read 4 bytes @ address 0x000000A2 (Data = 0x49034803)
Reading 64 bytes @ address 0x00000080
Read 4 bytes @ address 0x000000A2 (Data = 0x49034803)
Read 4 bytes @ address 0x20000534 (Data = 0x00000400)
```
Am I configuring something wrongly?

### Versions

* [plug-in version]
[image]
* [Eclipse version] ```Eclipse IDE for C/C++ Developers (includes Incubating components) Version: 2022-06 (4.24.0) Build id: 20220609-1112```
* [Java version] ```java version "1.8.0_201" Java(TM) SE Runtime Environment (build 1.8.0_201-b09) Java HotSpot(TM) 64-Bit Server VM (build 25.201-b09, mixed mode)
```
* [operating system] Windows 10

Please understand that without being able to reproduce the bug we cannot identify your problem.

---

ilg-ul, 2022-09-06:
> Am I configuring something wrongly?

It is hard to say what is wrong.

Can you also show the content of the **Debugger** tab?

As a general observation, although writing the Reset_Handler in C is theoretically possible, in certain conditions modern compiler do some optimisations and the resulting code is not functional.

ilg-ul, 2022-09-06:
Another recommendation, when debugging C code fails, switch to assembly mode, and try to understand the generated code.

XaviLC, 2022-09-06:
> It is hard to say what is wrong.
>
> Can you also show the content of the **Debugger** tab?
>
> As a general observation, although writing the Reset_Handler in C is theoretically possible, in certain [...]

What debugger tab?

ilg-ul, 2022-09-06:
> What debugger tab?

The one between Main and Startup.

XaviLC, 2022-09-06:
> > What debugger tab?
> 
> The one between Main and Startup.

Sorry I don't know what you mean, I have put in the ticket the debugger stack already, that is the first thing that happens after flashing the board

TommyMurphyTM1234, 2022-09-06:
@ilg-ul is referring to the _Debugger_ tab in the debug launch configuration GUI.

https://rtist.hcldoc.com/help/index.jsp?topic=%2Forg.eclipse.cdt.doc.user%2Fgetting_started%2Fcdt_w_debug.htm

TommyMurphyTM1234, 2022-09-06:
> ### Description
> Setting a break point on Reset_Handler seems to stall at fpu init and not going to main using Jlink with cortex m7, however if I specify the program counter at the right address of the main , then I see the starting up going through. 

What does "specify the program counter at the right address of main" mean?

Do you mean that your program gets doesn't execute correctly if you put an initial breakpoint at `Reset_Handler` but does if you put the initial breakpoint at `main()`?

Is the whole program in and executing from flash?
Is there any relocation/copying of code from flash to RAM during the program initialization?

XaviLC, 2022-09-06:
> @ilg-ul is referring to the _Debugger_ tab in the debug launch configuration GUI.
> 
> https://rtist.hcldoc.com/help/index.jsp?topic=%2Forg.eclipse.cdt.doc.user%2Fgetting_started%2Fcdt_w_debug.htm

Ok, clear, thanks!
[image]

XaviLC, 2022-09-06:
> What does _"specify the program counter at the right address of main"_ mean?

It means that if I go to `Startup` >> `Ru/Restart Commands` and I fill in the box of `Set Program counter at (hex)` with the program counter of my application, which I obtain looking at the assembly code addresses; then, after rebooting I see who the debug trace shows the program breaking at the _main_ as I expect.

> Do you mean that your program doesn't execute correctly if you put an initial breakpoint at `Reset_Handler` but does if you put the initial breakpoint at `main()`?

Nope, in both cases the the inital breakpoint is not pointing to the right code section, it is breaking at `Target halted (DBGRQ, PC = 0x00011F9C)` which is the address of the init of the FPU in my cortex-m7

> Is the whole program in and executing from flash? If you're flashing the program using some flash tool before debugging and then debugging it from flash then why do you have the debug launch configuration loading the executable to the target at debug startup? Is there any relocation/copying of code from flash to RAM during the program initialization? Is `Reset_Init` your only startup code? Is there nothing else? For example to initialize BSS/zero init memory, heap (if used), copy static initialized data etc.?
> 
> You need to provide more details and info about your project and debug launch configuration etc. in order for others to offer useful suggestions.

TommyMurphyTM1234, 2022-09-06:
You haven't explained important details about your program that I asked about. Flash Vs RAM memory layout, relocation/copying, startup code etc. Without detailed info we're all just guessing.

XaviLC, 2022-09-06:
> > the debugger seems to stall in the fpu init
> 
> What does this mean? Did you single step the assembly code?

Not needed, as you see in the debugger trace the address is already shown there
[image]

XaviLC, 2022-09-06:
> You haven't explained important details about your program that I asked about. Flash Vs RAM memory layout, relocation/copying, startup code etc. Without detailed info we're all just guessing.

I mentioned above that I was using Flash and I have shared the scatter file I use.

> E.g. why are you using hex PC values for breakpoints instead of symbolic names such as `main`? That doesn't make sense to me.

This is exactly the point, I want to use the symbolic name `main`, but after the .elf is loaded on the M7, this main is not pointing to my expected main which is defined in `reset handler ` function that I shared but it starts at the FPU init

> And, as ever, creating a minimal reproducible test case that you could just share in full (whole project) would be ideal.

ilg-ul, 2022-09-06:
> Not needed, as you see in the debugger trace the address is already shown there

If you think that the address is enough, then you should be able to figure yourself why the program crashed.

TommyMurphyTM1234, 2022-09-06:
> > > the debugger seems to stall in the fpu init
> > 
> > 
> > What does this mean? Did you single step the assembly code?
> 
> Not needed, as you see in the debugger trace the address is already shown there [image]

What happens if you debug/single instruction step from the very start of your program? That is, don't use a breakpoint at all and don't let the debugger run the program at startup.

If your symbolic info is not matching the program in memory then you probably have some issue with the project build itself.

Again it's difficult to comment usefully with only bits of info.

XaviLC, 2022-09-07:
> What happens if you debug/single instruction step from the very start of your program? That is, don't use a breakpoint at all and don't let the debugger run the program at startup.
>
> If your symbolic [...]

Thanks for the pointers, I will take a look into the reset_init as you mentioned above. I am sure that the project build is fine and the image too, because I have tried exactly same build environment and .elf with Lauterbach + Lauterbach eclipse plugin and there it works correctly. So it breaks at main when I set the break point there at start

TommyMurphyTM1234, 2022-09-07:
Just to recap, things that I would look at:

1. If the program is being programmed to flash before debugging then remove the load of the program to the target from the debug launch configuration. At best it's unnecessary but at worst it could cause problems if the debugger attempts to do normal memory writes to a flash area of the memory map.
2. Check the symbolic info using a map or list file or even by just printing using the GDB console interface. For example, `p/x main` and `p/x Reset_Handler`.
3. Remove the continue from the debug launch configuration, don't set any breakpoint, and single step from the very start to see where/how/why execution fails.
4. Double check that, if you're debugging the program from flash, that the debugger is not trying to use soft breakpoints (where it saves the breakpoint instruction and temporarily overwrites it with a `BKPT` instruction - something that will generally not work directly with flash) but is using hardware breakpoints.
5. Check all debugging logs for any errors/anomalies.

XaviLC, 2022-09-07:
> Just to recap, things that I would look at:
>
>     1. If the program is being programmed to flash before debugging then remove the load of the program to the target from the debug launch [...]

How/where to check the symbolic info if not using GDB print? Via assembly step?

Thanks @TommyMurphyTM1234 for all the suggestions!

TommyMurphyTM1234, 2022-09-07:
> Do you mean in here? [image]

Yes. If the program is already in flash and being run/debugged from there then the debugger should not be trying to load it to the target. Uncheck the _Load executable_ option.

> How/where to check the symbolic info if not using GDB print? Via assembly step?

As I said already, list/map files, and maybe also using the `nm` binutil (if you're using a GCC toolchain - other toolchains should have an equivalent tool).

XaviLC, 2022-09-15:
Hi @TommyMurphyTM1234 @ilg-ul  one question, where/how does Eclipse Embedded set the stack Pointer?
I am seeing this but that address is invalid 
[image]
since my address ranges are
```XML
<MemMap>
  <CoreRunning>
      <Range Start="0x00000000" Size="0x20000" Zone="Default"/>
      <Range Start="0x20000000" Size="0x7FF8" Zone="Default"/>
    </CoreRunning>
    <CoreHalted>
      <Range Start="0x00000000" Size="0x20000" Zone="Default"/>
      <Range Start="0x20000000" Size="0x7FF8" Zone="Default"/>
    </CoreHalted>
</MemMap>
```

TommyMurphyTM1234, 2022-09-15:
> Hi @TommyMurphyTM1234 @ilg-ul one question, where/how does Eclipse Embedded set the stack Pointer?

It doesn't. Cortex-M do this automatically.
At startup/post reset they load $sp with $VTOR[0] and $pc with $VTOR[4].
Some (early?) cores don't have a $VTOR in which case it's effectively 0x00000000.

VTOR = Vector Table Offset Register.

ilg-ul, 2022-09-15:
> Some (early?) cores don't have a $VTOR in which case it's effectively 0x00000000

ARM-v6m (Cortex-M0/M0+) cores do not implement `VTOR`.

TommyMurphyTM1234, 2022-09-15:
> > Some (early?) cores don't have a $VTOR in which case it's effectively 0x00000000
> 
> ARM-v6m (Cortex-M0/M0+) cores do not implement `VTOR`.

Thanks @ilg-ul - couldn't remember offhand. 🙂

XaviLC, 2022-09-15:
> > Hi @TommyMurphyTM1234 @ilg-ul one question, where/how does Eclipse Embedded set the stack Pointer?
> 
> It doesn't. Cortex-M CPUs do this automatically. At startup/post reset they load `$sp` with `$VTOR[0]` and `$pc` with `$VTOR[4]`. Some (early?) cores don't have a `$VTOR` in which case it's effectively `0x00000000`.
> 
> VTOR = Vector Table Offset Register.
> 
>     * https://developer.arm.com/documentation/ka001146/latest
> 
>     * https://embetronicx.com/tutorials/microcontrollers/stm32/reset-sequence-in-arm-cortex-m4/amp/

Thanks! 
And what happens if the application is only in RAM is the same?

ilg-ul, 2022-09-15:
> what happens if the application is only in RAM

The debugger starts from the entry point, usually defined in the linker script.

XaviLC, 2022-09-15:
So if my .ld file has this 
```/* Linker script to configure memory regions.
 * Need modifying for a specific board.
 *   ITCM.ORIGIN: starting address of flash
 *   ITCM.LENGTH: length of flash
 *   DTCM.ORIGIN: starting address of RAM bank 0
 *   DTCM.LENGTH: length of RAM bank 0
 */
MEMORY
{
  ITCM (rx) : ORIGIN = 0x0, LENGTH = 0x20000 /* 128K */
  DTCM (rwx) : ORIGIN = 0x20000000, LENGTH = 0x7FF8 /* 8K */
}

/* Linker script to place sections and symbol values. Should be used together
 * with other linker script that defines memory regions ITCM and DTCM.
 * It references following symbols, which must be defined in code:
 *   Reset_Handler : Entry of reset handler
 *
 * It defines following symbols, which code can use without definition:
 *   __exidx_start
 *   __exidx_end
 *   __copy_table_start__
 *   __copy_table_end__
 *   __zero_table_start__
 *   __zero_table_end__
 *   __etext
 *   __data_start__
 *   __preinit_array_start
 *   __preinit_array_end
 *   __init_array_start
 *   __init_array_end
 *   __fini_array_start
 *   __fini_array_end
 *   __data_end__
 *   __bss_start__
 *   __bss_end__
 *   __end__
 *   end
 *   __HeapLimit
 *   __StackLimit
 *   __StackTop
 *   __stack
 */
ENTRY(Reset_Handler)
...
```
It should go to the Reset_Handler right?

ilg-ul, 2022-09-15:
> I don't understand why the hack the SP is pointing to that value

Because your program crashed.

TommyMurphyTM1234, 2022-09-15:
> > > It should go to the `Reset_Handler` right?
> > 
> > 
> > That's correct.
> 
> Then I don't understand why the hack the SP is pointing to that value XD
> 
> EDIT: I see that as parameters to GDB client setup by default I have `set mem inaccessible-by-default off` shouldn't this be set to `set mem inaccessible-by-default on`?

No. That allows the debugger to access memory regions that are not defined in the linker script. If you set that on then you'd have to describe the full memory map, including memory mapped peripherals etc., in the linker script. It's very common for this to be set to off for embedded targets.

TommyMurphyTM1234, 2022-09-15:
> > I don't understand why the hack the SP is pointing to that value
> 
> Because your program crashed.

And I still don't know why you still haven't tried single asm instruction debugging from reset to see where things go wrong as I suggested several times.... :⁠-⁠|

XaviLC, 2022-09-16:
> > > I don't understand why the hack the SP is pointing to that value
> > 
> > 
> > Because your program crashed.
> 
> And I still don't know why you still haven't tried single asm instruction debugging from reset to see where things go wrong as I suggested several times.... :⁠-⁠|

I have, and that is why I am asking about the SP, because as soon as I boot I see the SP already wrong as shown in the pic above 
[image]
So now I am trying to understand where this wrong state comes from...

ilg-ul, 2022-09-16:
> where this wrong state comes from...

If your program runs entirely from RAM, you have to manually set SP, since the automatic setting from the interrupts table does not occur.

TommyMurphyTM1234, 2022-09-16:
> > where this wrong state comes from...
> 
> If your program runs entirely from RAM, you have to manually set SP, since the automatic setting from the interrupts table does not occur.

The target is Cortex-M7 I believe? That being the case, I would initialise the `$VTOR` via gdb commands in the debug launch configuration.

`set $VTOR=0x20000000`

TommyMurphyTM1234, 2022-09-16:
> > where this wrong state comes from...
> 
> If your program runs entirely from RAM, you have to manually set SP, since the automatic setting from the interrupts table does not occur.

The target is Cortex-M7 I believe? That being the case, I would initialise the `$VTOR` via gdb commands in the debug launch configuration.

`set *0xE000ED08=0x20000000`

TommyMurphyTM1234, 2022-09-16:
> I have, and that is why I am asking about the SP, because as soon as I boot I see the SP already wrong as shown in the pic above [image] So now I am trying to understand where this wrong state comes from...

If you're compiling your program for RAM at `0x20000000`, and running/debugging it from there, then that screenshot suggests that your vector table is not correctly set up. It shows `0x20000000` containing `0x1` and that is not a valid initial `$sp` value.

TommyMurphyTM1234, 2022-09-16:
You might be better off taking a step back and using a known good example project and debug launch configuration to see how compiling for and running/debugging from RAM is done, and then seeing where your own project and debug launch configuration deviates from this.

ilg-ul, 2022-09-16:
If you update the Eclipse Embedded CDT to v6.3, you'll find a new Hello World QEMU project template, which generates functional projects for Arm and RISC-V devices that run on QEMU.

ilg-ul, 2023-02-06:
@XaviLC, did you identify the problem with your project?

XaviLC, 2023-03-20:
> @XaviLC, did you identify the problem with your project?

Hi @ilg-ul no not yet, I am still facing the same issue which I don't see it with for instance Ozone IDE from Segger, :(

TommyMurphyTM1234, 2023-03-20:
> > @XaviLC, did you identify the problem with your project?
> 
> Hi @ilg-ul no not yet, I am still facing the same issue which I don't see it with for instance Ozone IDE from Segger, :(

Did you do as I suggested here?

* https://github.com/eclipse-embed-cdt/eclipse-plugins/issues/540#issuecomment-1249273494

ilg-ul, 2023-03-20:
I did not try clang for embedded projects yet, so I don't know how well it is supported by GDB.

If you discover how to make it work, please share your findings.

XaviLC, 2023-03-21:
> I did not try clang for embedded projects yet, so I don't know how well it is supported by GDB.
> 
> If you discover how to make it work, please share your findings.

Well then this could be an issue that I am facing

TommyMurphyTM1234, 2023-03-21:
> 1. The compiler I am using is clang so I am trying ensure whether the gdb I am using (GNU) is fully compatible or not

I don't understand.
What sort of compatibility are you referring to here?
As long as the compiler/linker generates an ELF file containing appropriate symbolic debug information (presumably DWARF or some variant) then there should be no issue?

XaviLC, 2023-03-22:
So I do know that with gcc it works, but my project is clang based so I need to use it now matter what :). That is why I am suspecting from the .S files because those are different for clang and gcc.

ilg-ul, 2023-03-22:
> ... `.S` files are different for clang and gcc

What is different? Does clang use a different assembler, with a different syntax?

XaviLC, 2023-03-22:
> > ... `.S` files are different for clang and gcc
> 
> What is different? Does clang use a different assembler, with a different syntax?

Yes it does, and I cannot find so far an example of that. For instance entries like `.arch` are not supported any more or not like that

ilg-ul, 2023-03-22:
The source files used to compile the toolchain should have startup files in assembly. Also taking a look at the assembler source code, the Cortex-M specific files, might help, perhaps they have some explanations. Or even some test files.

closed 2023-04-26 by ilg-ul