eclipse-embed-cdt/eclipse-plugins #543: I was following the "blinky-arm" tutorial with the installation of all "latest" tools. However, there are build errors as shown below:

ilg-ul, 2022-10-09:
I was following the "blinky-arm" tutorial with the installation of all "latest" tools. However, there are build errors as shown below:

"""
Building file: ../system/src/newlib/syscalls.c
Invoking: GNU Arm Cross C Compiler
arm-none-eabi-gcc -mcpu=cortex-m4 -mthumb -mfloat-abi=soft -Og -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-move-loop-invariants -Wall -Wextra -g3 -DDEBUG -DUSE_FULL_ASSERT -DOS_USE_SEMIHOSTING -DTRACE -DOS_USE_TRACE_SEMIHOSTING_DEBUG -DSTM32F407xx -DUSE_HAL_DRIVER -DHSE_VALUE=8000000 -I"../include" -I"../system/include" -I"../system/include/cmsis" -I"../system/include/stm32f4-hal" -std=gnu11 -MMD -MP -MF"system/src/newlib/syscalls.d" -MT"system/src/newlib/syscalls.o" -c -o "system/src/newlib/syscalls.o" "../system/src/newlib/syscalls.c"
../system/src/newlib/syscalls.c: In function 'findslot':
../system/src/newlib/syscalls.c:599:26: error: 'struct _reent' has no member named '__sdidinit'
  599 |       if ((ptr) && !(ptr)->__sdidinit)          \
      |                          ^~
../system/src/newlib/syscalls.c:613:3: note: in expansion of macro 'CHECK_INIT'
  613 |   CHECK_INIT(_REENT);
      |   ^~~~~~~~~~
make: *** [system/src/newlib/subdir.mk:39: system/src/newlib/syscalls.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

21:05:10 Build Failed. 2 errors, 0 warnings. (took 4s.805ms)
"""

I assume this may be related to the incompatible versions of the tools. How to handle this?

_Originally posted by @eraus in https://github.com/eclipse-embed-cdt/eclipse-plugins/discussions/542#discussioncomment-3831159_

ilg-ul, 2022-10-09:
With the new versions of newlib, the macro should read `(ptr)->__cleanup)`, not `(ptr)-> __sdidinit)`.

ilg-ul, 2022-10-09:
Fixed on 2022-10-09.

eraus, 2022-10-09:
I have just tried on Eclipse Marketplace. The newest version is still 6.3.0. I will wait a bit until it appears on the Marketplace to give it a try. Many thanks!

ilg-ul, 2022-10-09:
Ah, sorry for the misunderstanding, the fix is only in the project git, it was not published yet, due to a problem with the GitLab build infrastructure.

I'll make the release when possible and let you know.

eraus, 2022-10-09:
Great! Thanks!

ilg-ul, 2022-10-09:
As a workaround, you can also manually patch that faulty macro, it is a simple change.

eraus, 2022-10-09:
Sounds good. I see your diffs.

ilg-ul, 2022-10-10:
6.3.1 was published.

Please check it and confirm it also works in your environment.

eraus, 2022-10-10:
I see 6.3.1 published. Here are what I have tried. 

1. When I went to the Eclipse Marketplace, I see my 6.3.0 was automatically updated to 6.3.1. As I am a new user for Eclipse, I am not sure if this was done automatically or not. To make sure, I removed it and reinstalled 6.3.1.
2. The xPack tools I installed was recognized with the new 6.3.1.
3. I then created a new Blinky project, I see that the `syscalls.c` still has `(ptr)->__sdidinit)`. 
4. I have not patched up the files yet. I am still in the process trying to figure out where the original files are saved. Before installing 6.3.1, I thought the original `syscalls.c` file was part of 6.3.1. Now that you have fixed the issue in the repo, but I still have this issue, I am not sure where it came now. Please shed some lights here, and I will do some more tries after this.

ilg-ul, 2022-10-10:
I'm afraid you messed something :-(

I suggest you download a fresh Eclipse Embed CDT, create a new workspace and install the plug-ins.

The normal way is to simply update the Eclipse, but once something goes wrong, sometimes there is not much to do to fix it than restart from scratch.

eraus, 2022-10-10:
I started a new workspace, and everything worked well with the newest versions of tools. Many thanks!

ilg-ul, 2022-10-10:
Thank your help!

closed 2022-10-10 by ilg-ul