eclipse-embed-cdt/eclipse-plugins #551: Existing projects using old toolchain id shows error after upgrade

vishnusarathashling, 2023-02-04:
### Description
Existing projects (created using previous embedCDT) using old toolchain id shows error after imported to IDE updated with new embed cdt (6.3.1). When importing exiting projects  The toolchain used was "RISC-V GCC/Newlib". And the toolchain path defined (via ini file) is `org.eclipse.embedcdt.managedbuild.cross.riscv.core/toolchain.path.2032619395=${eclipse_home}/../toolchain/riscv64-unknown-elf/bin` And the .cproject file is having the toolchain id options saved as :

`<option id="ilg.gnumcueclipse.managedbuild.cross.riscv.option.toolchain.id.437584809" superClass="ilg.gnumcueclipse.managedbuild.cross.riscv.option.toolchain.id" useByScannerDiscovery="false" value="-2032619395" valueType="string"/>`

Earlier I hope we were using the toolchain.name option and it's hash for identifying the path it was working fine. But now when we changed that to use toolchain.id we are not able to detect the path configured. When I tried to debug the issue I found like **org.eclipse.embedcdt.managedbuild.cross.core.preferences.PersistentPreferences::getKeyId** (from line 109 `keyId = Long.parseLong(toolchainId.trim());`) is returning the value directly from the option (in .cproject) which will be negative value (-2032619395 saved in existing project). Would it be better to change that to return Absolute value (like how we are mapping toolchain name to id). I tried changing the line to `keyId = Math.abs(Long.parseLong(toolchainId.trim()));` and it's working fine. Is it good for us to keep this so as to have the backward compatility?

vishnusarathashling, 2023-02-04:
Platform used:
OS: Win 10
Eclipse: 2022-09
Eclipse CDT : 10.7.1
Eclipse Embedded CDT : 6.3.1

vishnusarathashling, 2023-02-04:
@ilg-ul Please share your valuable thoughts on this.

ilg-ul, 2023-02-04:
The story is old, and indeed the rule to identify toolchains has a problem and was changed when preparing the first embedCDT release.

> I tried changing the line to keyId = Math.abs(Long.parseLong(toolchainId.trim())); and it's working fine.

If you still have old projects and this change allows to import them, sure, we can consider it. The only concern is to not affect other projects. Are you sure that new projects, created with embedCDT, cannot have negative values there?

vishnusarathashling, 2023-02-04:
Please see my understanding, correct me if I am wrong.
* The toolchain id option value (in .cproject) is taken from the id value given for the toolchain in ToolchainDefinition (for ARM and RISCV) or from the extentions of the extension points `org.eclipse.embedcdt.managedbuild.cross.arm.core.toolchains`, `org.eclipse.embedcdt.managedbuild.cross.riscv.core.toolchains` (or custom definition toolchains).
* If someone has created a new toolchain with negative id, then that will be reflected in the .cproject.
* But the function where we are proposing the change is normally called for getting the key to be used for preference look up. As you told if someone has created a key like `toolchain.path.-<id>`, starting with a negative, this change can create issue for new projects. Yeah I got the issue. But I thought none will be adding that kind of a key.

ilg-ul, 2023-02-04:
I hope that negative keys are not in use, but, since they were valid at the time they were computed from hashes, there is no way to guarantee it.

For old projects that throw this error, can you later fix them by manually selecting the new toolchain?

vishnusarathashling, 2023-02-04:
I think even though negative toolchain id's were valid earlier (when we used Hash), the keys were always non-negative since we were using the below function 
[image]
which used the abs value as key always.

But I think with our new approach it supports negative keys in preference.

> For old projects that throw this error, can you later fix them by manually selecting the new toolchain?
Yes after import if we are changing the toolchain manually the .cproject is getting updated with correct (new toolchain) and everything working fine.

May be I will just try to have a call to PersistentPreferences::getToolchainPath(String toolchainName, IProject project) [as a second level call for backward compatibility] [the current deprecated function] if we are not getting the path resolved using toolchainId and toolchainName. I will add this change for my product, since we have a quite number of existing projects which has to be supported :(.

ilg-ul, 2023-02-04:
If you find a solution that fixes the problem without affecting possible existing projects which use negative values, please let me know.

vishnusarathashling, 2023-02-04:
The way I see if we are using this `PersistentPreferences::getToolchainPath(String toolchainName, IProject project)` as a fallback method if we are not getting toolchain path with toolchainId and toolchainName, I think the toolchain path resolution for both the new and exiting projects will be addressed (the sol I mentioned above). But I am not sure  it would be accepted at a framework level (calling a deprecated method as a fallback). Anyways I will create a PR with the changes, so you can have a keen review.

ilg-ul, 2023-02-04:
> But I am not sure it would be accepted at a framework level (calling a deprecated method as a fallback)

Well, it is not ideal, but, if it fixes the problem without affecting compatibility, I think we can accept it.

vishnusarathashling, 2023-02-06:
> > But I am not sure it would be accepted at a framework level (calling a deprecated method as a fallback)
> 
> Well, it is not ideal, but, if it fixes the problem without affecting compatibility, I think we can accept it.

ok @ilg-ul Please see through the PR : https://github.com/eclipse-embed-cdt/eclipse-plugins/pull/552.

ilg-ul, 2023-02-06:
Merged, thank you.

I'll delay the new release a few more days, in case you have further contributions.

closed 2023-02-06 by ilg-ul

vishnusarathashling, 2023-02-06:
> Merged, thank you.
> 
> I'll delay the new release a few more days, in case you have further contributions.

Thanks @ilg-ul for the support.