eclipse-embed-cdt/eclipse-plugins #554: Arm project templates generate linker warning "<project> has a LOAD segment with RWX permissions"

TommyMurphyTM1234, 2023-02-16:
Tools used

1. Eclipse IDE for Embedded C/C++ Developers (includes Incubating components) Version: 2022-12 (4.26.0) Build id: 20221201-1913
2. arm-none-eabi-gcc (xPack GNU Arm Embedded GCC x86_64) 12.2.1 20221205

Create a new project using the "Hello Workd Arm Cortex-M C/C++ Project" and then compile it.
The following warning is generated:
```
/home/tommy/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: warning: arm-hello-world.elf has a LOAD segment with RWX permissions
```
It may be innocuous and not unexpected but I can't see it mentioned anywhere so am logging it in case it needs attention.

I think that this happens with any Arm template project.

ilg-ul, 2023-02-16:
Yes, I already entered a ticket asking to disable this warning for embedded toolchains:

- https://bugs.linaro.org/show_bug.cgi?id=5915

ilg-ul, 2023-02-16:
There is nothing to be done by the plug-ins to fix, we can close it.

closed 2023-02-16 by ilg-ul

TommyMurphyTM1234, 2023-02-16:
That's fine @ilg-ul - thanks.