eclipse-embed-cdt/eclipse-plugins #559: Remove references Navigator view to support 2023-06

ruspl-afed, 2023-04-26:
### Steps to Reproduce

1. Configure Eclipse CDT with Oomph (with Eclipse Platform 2023-06 M1+ in .target)
2. Import sources from https://github.com/eclipse-embed-cdt/eclipse-plugins , `master` branch 
3. Wait for workspace build to be completed

**Expected behaviour:** No compilation errors

**Actual behaviour:** `ID_RES_NAV cannot be resolved or is not a field`

ruspl-afed, 2023-04-26:
Hello @ilg-ul 
The Navigator and related code has been [removed](https://github.com/eclipse-platform/eclipse.platform.ui/issues/644) from Eclipse Platform. It seems that we will need a service release to support SimRel 2023-06.

I can see milestone 6.3.2 without open issues, but it is not released yet. Should we add this issue to 6.3.2 as well or do you prefer to release the current state first?
cc: @jonahgraham

ilg-ul, 2023-04-26:
I guess we have to do this anyway, but, just for the record, what means `ID_RES_NAV`?

ruspl-afed, 2023-04-26:
This is the identifier of old good "Navigator" view, which has been [removed ](https://github.com/eclipse-platform/eclipse.platform.ui/issues/644) from Eclipse Platform starting from 2023-06 M1

ilg-ul, 2023-04-26:
I don't remember using this old Navigator view. Will this removal have any practical impact on the Embedded CDT functionality?

ruspl-afed, 2023-04-26:
I'm not sure how it was used it in the context of Embedded CDT functionality, perhaps this code was just generated by `org.eclipse.ui.perspectives` extension point template long ago.

ilg-ul, 2023-04-26:
> I'm not sure how it was used it in the context of Embedded CDT functionality

Neither do I :-(

> perhaps this code was just generated ...

Yes, it is possible. Or it was the result of a copy/paste from another plugin.

ruspl-afed, 2023-04-26:
The PR is ready for your review #560

AntonKrug, 2023-04-26:
https://stackoverflow.com/questions/60962386/why-is-eclipse-removing-the-navigator-view

AntonKrug, 2023-04-26:
I just used Project Explorer exclusively

jonahgraham, 2023-04-26:
We need to test that versions of embed cdt that are already released work when they hit that code and installed on 2023-06. It's just a constant, so probably inlined and fine. But it could lead to class not found exception at runtime.

ilg-ul, 2023-04-26:
@jonahgraham, I'm not sure what we should do next. 

I was about to prepare 6.3.2 based on the current `develop` branch, for testing. Is this ok?

jonahgraham, 2023-04-26:
Please go ahead on that.

We just need to test to find out if 6.3.2 will be minimum version **required** for all users wanting to use 2023-06.

ilg-ul, 2023-04-26:
The first build of 6.3.2 can be tested from <https://download.eclipse.org/embed-cdt/builds/develop/p2/>.

On my 2023-03 it seems ok.

Jonah, when you confirm that it is ok with 2023-06, I'll proceed to merge the commits into master and complete the release.

ilg-ul, 2023-04-27:
> I have now tested 6.3.1 in 2023-06 and it does not raise a Class not found exception

How did you do this? The definition was in the packs.ui plug-in, in one of the shortcuts there. I have no idea what `IPageLayout.ID_RES_NAV` means in that context, and how to test it.

jonahgraham, 2023-04-27:
Here is a screenshot from 2023-03:

[image]

The entries in the Show View menu are populated with `fFactory.addShowViewShortcut`

In 2023-06 I can see that the code (`addViewShortcuts`) is run, but the Show View menu is simply missing the Navigator entry.

Because the  `IPageLayout.ID_RES_NAV` constant was inlined we don't get any weird java exceptions because `ID_RES_NAV` was deleted.

jonahgraham, 2023-04-27:
PS that is the show view menu when you have the packs perspective open

ilg-ul, 2023-04-27:
Great, thank you for testing and for the explanation!

Tomorrow I'll proceed with the release.

ilg-ul, 2023-04-28:
6.3.2 is out.

jonahgraham, 2023-04-28:
Thanks @ilg-ul for doing the release and @ruspl-afed for keeping the code up to date with platform changes.