eclipse-embed-cdt/eclipse-plugins #563: The disassembly view dosen't show opcode

ljw66, 2023-05-17:
Describe the bug
When I debug a remote riscv project. I run gdb in cmd and input "disas /r" command. The output has opcodes. When I use the same gdb debug the same execute file. I open disassembly view and click show opcodes. The disassembly text move to the right a little but no opcode. In the gdb console, I write "disas /r" and the output contains opcode.

Screenshots
[image]
[image]
[image]
[image]

TommyMurphyTM1234, 2023-05-17:
Isn't that an option in the GUI to display the opcodes?

* https://mcuoneclipse.com/2015/07/22/show-opcodes-in-eclipse-disassembly-view/

ljw66, 2023-05-17:
> Isn't that an option in the GUI to display the opcodes?
> 
> * https://mcuoneclipse.com/2015/07/22/show-opcodes-in-eclipse-disassembly-view/

Thanks for your help. That's it, in the second image, I didn't click show opcodes and there is no opcodes. But in the third picture, I clicked the show opcodes option. The result is address column move right a little. But still no opcodes.

TommyMurphyTM1234, 2023-05-17:
What version of the various tools are you using as per the issue template?
> ### Versions
>
> * [plug-in version]
> * [Eclipse version]
> * [Java version]
> * [operating system]
> * [toolchain version]

ljw66, 2023-05-17:
eclipse version: 2022-12 (4.26.0)
plugin-in version: 6.3.1.202210101738
java version: 17
toolchain version: Xuantie-900-gcc-elf-newlib-mingw-V2.6.1
operating system: windows

TommyMurphyTM1234, 2023-05-17:
> toolchain version: Xuantie-900-gcc-elf-newlib-mingw-V2.6.1 

What happens if you use the xPack RISC-V GCC toolchain?

TommyMurphyTM1234, 2023-05-17:
> Due to the special chip used now, only this toolchain can be used for compilation. The xPack RISC-V GCC toolchain fail to support some comipile flags.

OK, but you need to determine if the issue is due to the toolchain rather than the Embedded CDT plugins.

ljw66, 2023-05-17:
Alright, thank you very much!

closed 2023-05-17 by ljw66

TommyMurphyTM1234, 2023-05-17:
I tried the following but neither see opcodes in the Disassembly view (when debugging a RISC-V RV64 program on QEMU) nor any option to show opcodes in the view menu or global preferences.

* Eclipse IDE for Embedded C/C++ Developers 4.27.0.20230309-1200
* Embedded C/C++ Core 6.3.2.202304281106

I also have these xPacks installed after creating a RISC-V template project:
```
tommy@gannet:~$ xpm list --global
- @micro-os-plus/architecture-riscv
  A source library xPack with the µOS++ RISC-V architecture port
  - 4.1.0
- @micro-os-plus/build-helper
  A source library xPack to assist the µOS++ builds
  - 2.11.0
- @micro-os-plus/devices-qemu-riscv
  A source xPack with <your-description-here>
  - 1.0.1
- @micro-os-plus/diag-trace
  A source library xPack with the µOS++ trace::printf() tracing infrastructure
  - 4.1.1
- @micro-os-plus/hello-world-qemu-template
  A source xPack  / npm package with a template to generate semihosted Hello World projects running on QEMU
  - 1.4.1
- @micro-os-plus/semihosting
  A source library xPack with the µOS++ portable semihosting definitions
  - 8.0.0
- @micro-os-plus/startup
  A source library xPack with the µOS++ startup code for bare-metal platforms
  - 5.2.0
- @xpack-dev-tools/arm-none-eabi-gcc
  A binary xPack with the GNU Arm Embedded GCC toolchain executables
  - 12.2.1-1.2.1
- @xpack-dev-tools/cmake
  A binary xPack with the CMake executables
  - 3.21.6-1.1
- @xpack-dev-tools/ninja-build
  A binary xPack with the Ninja Build executables
  - 1.11.0-1.1
- @xpack-dev-tools/qemu-arm
  A binary xPack with QEMU Arm executables
  - 7.2.0-1.1
- @xpack-dev-tools/qemu-riscv
  A binary xPack with QEMU RISC-V executables
  - 7.0.0-1.1
  - 7.2.0-1.1
- @xpack-dev-tools/riscv-none-elf-gcc
  A binary xPack with the GNU RISC-V Embedded GCC executables
  - 12.1.0-2.1
  - 12.2.0-3.1
```

TommyMurphyTM1234, 2023-05-17:
Apologies - I missed that you click on the address column in the Disassembly view in order to get the Show Opcodes option.

[image]

As you can see here, using the set of tools that I listed earlier - including the latest xPack RISC-V embedded GCC - the opcodes are displaying correctly.

So, I can only infer that the original issue is due to the use of the different toolchain:

> Xuantie-900-gcc-elf-newlib-mingw-V2.6.1

TommyMurphyTM1234, 2023-05-17:
> Xuantie-900-gcc-elf-newlib-mingw-V2.6.1

Yes - good point.

ljw66, 2023-05-18:
I compared disassembled code generated by different toolchains. I found out that my toolchain uses big endian mode. But eclipse only show little endian. This makes it impossible to display the operand. Thank for everyone's suggestions.

ilg-ul, 2023-05-18:
> I found out that my toolchain uses big endian mode. 

What a regrettable design :-(

> But eclipse only show little endian.

I don't remember doing anything special in Eclipse Embedded CDT for this, the behaviour is probably inherited from CDT.

ljw66, 2023-05-18:
I think so. Thanks again :）