eclipse-embed-cdt/eclipse-plugins #566: CMSIS Packs - nomenclature is inconsistent and a little confusing

TommyMurphyTM1234, 2023-05-23:
Originally reported here:

* https://github.com/eclipse-embed-cdt/eclipse-plugins/discussions/518#discussioncomment-5977154

To summarise:

1. Should "CMSIS Packs" rather than also "CMSIS packs" etc. be used consistently?
2. The CMSIS Packs view initially says "(no packages) - Press the Refresh button to register all available packages" but the relevant button in the view toolbar (left arrow above right arrow) is called "Update the CMSIS packs definitions" according to the tooltip.
3. The use of terms "refresh", "register" and "update" are a bit confusing. I think that the "Update the CMSIS packs definitions" button causes the master index (or indices if there are multiple CMSIS Packs repositories configured?) to be downloaded so that thereafter the CMSIS Packs view has a local list of CMSIS Packs and where to find them etc.? CMSIS Packs are not themselves downloaded unless/until the user explicitly clicks on them and selects _Install_? That being the case the nomenclature should probably more accurately describe/reflect the operations that take place and should be consistent.

ilg-ul, 2023-05-23:
> 1. Should "CMSIS Packs" rather than also "CMSIS packs" etc. be used consistently? 

This first one was easy, fixed.

ilg-ul, 2023-05-23:
For 2 & 3, could you suggest how to update the messages?

TommyMurphyTM1234, 2023-05-23:
> For 2 & 3, could you suggest how to update the messages?

The problem I have is that I don't fully understand what's happening in this context to address the nomenclature issues. My assumptions are outlined above but I don't know if they're correct. These details need to be ascertained in other to use the appropriate nomenclature, verbs, nouns etc.

ilg-ul, 2023-05-26:
When you have time to suggest better messages, I'll update them and proceed with a new release.

ilg-ul, 2023-06-27:
@TommyMurphyTM1234, a new release is planned in a few days, probably the end of the week, so, if you have suggestion on how to improve the messages, now would be a good moment to contribute them.

ilg-ul, 2023-07-24:
@TommyMurphyTM1234, if there are no further suggestions, can you close this issue?