eclipse-embed-cdt/eclipse-plugins #567: Support enhanced GNU Tool Factory capabilities introduced in CDT 11.2.0

jld01, 2023-05-23:
eclipse-cdt/cdt#361 introduces improved support for launching GNU tools such as `objdump` where these tools are not on the global PATH and may have prefixed names such as `arm-none-eabi-objdump`.

The CDT Managed Build GNU toolchain descriptions provided by _Embedded CDT_ can take advantage of these new capabilites by:
- Modifying the toolchain descriptions to use the GNU ELF binary parser
- Providing the appropriate GNU tool prefix via a CDT build macro named `gnu_tool_prefix`

ilg-ul, 2023-05-23:
Thank you, John, contributions are welcome.

However it might take some time for Embedded CDT to move to 11.2, since the current plan is to remain compatible with older versions as much as possible.

jld01, 2023-05-23:
> However it might take some time for Embedded CDT to move to 11.2, since the current plan is to remain compatible with older versions as much as possible.

Understood. These changes are API-compatible with your current target definition and will have no effect (other than providing an unused CDT build variable) when deployed against CDT versions preceding 11.2.0 M2.

ilg-ul, 2023-05-23:
If the changes are only internal and do not result in incompatible project configuration, I guess they are ok.

@jonahgraham, any thoughts on this? Can you confirm that we'll not introduce any compatibility issues with projects created before?

jonahgraham, 2023-05-23:
@ilg-ul this change should not impact old projects negatively.

ilg-ul, 2023-05-23:
Ok, thanks for confirming. Merged.

I'll update plug-ins versions shortly.

PR #568 (2023-05-23): Provide GNU tool prefix for CDT GNU tool factory
Resolves #567
commit 62a35fac (2023-05-23): Provide GNU tool prefix for CDT GNU tool factory

Resolves #567
closed 2023-05-23 by ilg-ul