eclipse-embed-cdt/eclipse-plugins #569: Feature request: add option to connect to a different OpenOCD port

MicBiso, 2023-06-09:
### Description

In the Debugger tab I can choose the GDB, Telnet and Tcl ports, this is fine and it configures the ports when OpenOCD is invoked.
However I cannot choose a different port GDB number but the one that is specified in the option above.
The use case is related to multicore devices with different (and consecutive ports) for the different cores, e.g. core 1 port 3333, core 2 port 3334, and so on.
At the moment I can only connect to the core 1, i.e. port 3333 and if I specify 3334 as GDB port, it will connect to port 3334 but this will still be the 1st core, with 2nd core having port 3334 and so on. 
If I deselect the flag "Start OpenOCD locally" then for sure I can specify the exact port I want GDB to connect to but with the hassle to start OpenOCD manually.

It would be nice to have the possibility to specify the actual GDB port.
I tried to use -ex "target extended-remote localhost:3334" as "other options" but it does not work.

### Versions

* [plug-in version] 6.3.2
* [Eclipse version] 2022-06 (4.24.0)
* [Java version]
* [operating system] Ubuntu 22.04
* [toolchain version] NA

TommyMurphyTM1234, 2023-06-09:
Your report seems confusing.

This:

> In the Debugger tab I can choose the GDB, Telnet and Tcl ports, this is fine and it configures the ports when OpenOCD is invoked. 

seems to contradict this:

> However I cannot choose a different port GDB number but the one that is specified in the option above. 

> if I specify 3334 as GDB port, it will connect to port 3334 but this will still be the 1st core

Surely this behaviour is outside the scope of the plugin and dictated by something external?
Maybe you need a custom OpenOCD script in order to discriminate between different core targets?

I think you need to explain the issue in more detail because I cannot see that it's obviously an issue with the plugin itself.

TommyMurphyTM1234, 2023-06-09:
What OpenOCD script are you using to interact with your target?

MicBiso, 2023-06-09:
What do you mean by script? The configuration file?

[image]

MicBiso, 2023-06-09:
[image]

In other words, the "Port Number" here should be left configurable, also when the "Start OpenOCD locally" is checked.
In fact if I change the GDB port, the same is reflected there:

[image]

jonahgraham, 2023-06-09:
I notice you are using Renesas - I think Renesas' [e2studio](https://www.renesas.com/us/en/software-tool/e-studio) contains launch configurations that allow such customizations, including selecting which cores to connect Eclipse to.

MicBiso, 2023-06-09:
No, that's not e2studio, it's "plain" Eclipse:

[image]

TommyMurphyTM1234, 2023-06-09:
What if you do this:

1. Uncheck _Start OpenOCD locally_
2. Change _Remote Target > Port number_ to 3334
3. Check _Start OpenOCD locally_

Does GDB now use 3334?

MicBiso, 2023-06-09:
> But, it seems, that when OpenOCD is started locally by the plugin then GDB always connects to GDB OpenOCD Debugging > Debugger > OpenOCD Setup > GDB Port and there is no option to get it to connect to that plus 1 or whatever.

Correct.

> What if you do this:
> 
> Uncheck Start OpenOCD locally
> Change Remote Target > Port number to 3334
> Check Start OpenOCD locally
> Does GDB now use 3334?

Yes.

TommyMurphyTM1234, 2023-06-09:
[image]

TommyMurphyTM1234, 2023-06-09:
> Yes.

Ok - so that's the solution?

MicBiso, 2023-06-09:
Suboptimal I would say.
If I use the plugin I do not want to start OpenOCD manually myself.

TommyMurphyTM1234, 2023-06-09:
OK - I'll have to let @ilg-ul or @jonahgraham comment.

jonahgraham, 2023-06-09:
> In other words, the "Port Number" here should be left configurable, also when the "Start OpenOCD locally" box is checked.

Yes - I think that would solve the problem. By default the port number should track the port in start openocd locally, but there should be a way to override it.

MicBiso, 2023-06-09:
> If you want a single openocd instance to listen on multiple ports, you configure the first launcher to start it, and the second one to do not start it, but use the second port defined in the Remote Target field.

Isn't this a bit nasty? I just want to connect to the second port without having to first start an instance of the plugin just for this. As said, it would be enough to have the "Port Number" field to be always configurable.

MicBiso, 2023-06-09:
Perfect summary, thanks!

ilg-ul, 2023-06-09:
Do you want to start one or two sessions?

If you want to start a single session, but not for the first core, you have to adjust your script to give you access to that core, and make openocd to listen on a single port, which you can configure in the plugin.

I have close to zero experience with openocd, so I have no idea how your script should be changed, but with J-Link, if I remember right, for multi core devices it was possible to start multiple GDB server instances, each with its port, configured for a separate core, and this simple use case was assumed for openocd too.
 
From Tommy's summary, I understand that you want to start openocd locally with a custom multi-port configuration, let it listen on multiple port, and do not disable the Remote Target fields (which prevents out of sync configurations for regular use cases), to allow the GDB client to connect to another port.

This can be done, but is it really a good solution?

TommyMurphyTM1234, 2023-06-09:
> Do you want to start one or two sessions?

They want to start one - that connects to the "second" core target (Cortex-M33) for which OpenOCD creates a GDB server on port 3334.

> If you want to start a single session, but not for the first core, you have to adjust your script to give you access to that core, and make openocd to listen on a single port, which you can configure in the plugin.

That's one way to do it but invasive to the default target script.

> This can be done, but is it really a good solution?

I don't think that it's a bad solution.

As @jonahgraham said above, by default _Remote Target > Port number_ should match _OpenOCD Setup > GDB port_, but the former is left editable (even if _OpenOCD Setup > Start OpenOCD locally_ is checked) and GDB is launched by the plugin using the former rather than the latter. That allows the flexibility that is needed here but doesn't impact anything else that I can think of.

In short the suggested changes to the plugin are:

1. Always have _Remote Target > Port number_ enabled/editable
2. Always launch GDB to connect to _Remote Target > Port number_ rather than _OpenOCD Setup > GDB Port_

MicBiso, 2023-06-09:
Thanks @TommyMurphyTM1234! 
I started typing my answer but yours was better than mine!

jonahgraham, 2023-06-09:
I prototyped up in PR #570 a fix for this that adds this checkbox:

[image]

Is this a good solution? Or a non-starter?

TommyMurphyTM1234, 2023-06-09:
> I prototyped up in PR #570 a fix for this that adds this checkbox:
> 
> [image]
> 
> Is this a good solution? Or a non-starter?

At the moment I'd be more inclined to do this:

1. _OpenOCD Setup > GDB port_ and _Remote Target > Port number_ default to 3333
3. Changes to _OpenOCD Setup > GDB port_ are reflected/applied to _Remote Target > Port number_
4. Changes to _Remote Target > Port number_ override whatever was previously there (and are not reflected back to _OpenOCD Setup > GDB port_)
5. _Remote Target > Port number_ is always enabled - even if _OpenOCD Setup > Start OpenOCD locally_ is enabled
6. GDB is always launched by the plugin using _Remote Target > Port number_ rather than (the de-facto situation when _OpenOCD Setup > Start OpenOCD locally_ is checked?) _OpenOCD Setup > GDB port_?

jonahgraham, 2023-06-09:
> If the default for the checkbox is always true, and the value is not stored persistently in the environment to be used as a default for the next launcher, it is ok.

Yes - that is how I did it. Checkbox is always checked on new launches (i.e. doesn't use `fPersistentPreferences`), and is set back to checked if you toggle the start gdbserver to on as well.

> 
> My concern is that creating 'regular' launchers may inadvertently inherit this less used configuration.

Not sure what this means - please expand unless I addressed your issue in the above paragraph.

jonahgraham, 2023-06-09:
> At the moment I'd be more inclined to do this:

I like that better too. Maybe a tooltip on _Remote Target > Port number_ to explain this too.

jonahgraham, 2023-06-09:
Before I refactor based on @TommyMurphyTM1234's suggestion I will wait for @ilg-ul's input.

TommyMurphyTM1234, 2023-06-09:
> > At the moment I'd be more inclined to do this:
> 
> I like that better too. Maybe a tooltip on _Remote Target > Port number_ to explain this too.

My only concern is about if/how any settings get persisted and applied to future launch configurations.
But the way that Eclipse plugins persist settings (or not) across instances has always been a mystery to me.

ilg-ul, 2023-06-09:
> creating 'regular' launchers may inadvertently inherit this less used configuration

The mechanism of keeping persistent preferences, used by Eclipse in many places, is both helpful and annoying.

The annoying case would be when the user configures the remote target port 3334, runs a debug session, and the next day creates a regular launcher, for a simple device which starts openocd to listen only on 3333, and scratches his head why the plugin timeouts (and asks for support!).

Thus I would favour a solution which explicitly shows in the interface that the Remote Target is out of sync with the locally started server, and be sure this checkbox always defaults to a position that prevents this case.

ilg-ul, 2023-06-09:
> the way that Eclipse plugins persist settings (or not) across instances has always been a mystery to me

The mystery is not that deep, most values in the graphical interface are saved as persistent preferences and used as defaults when the plugin creates a new instance of the corresponding object. 

In this case, a subsequent OpenOCD launcher will inherit most of the values of the previously created OpenOCD launcher, a subsequent QEMU launcher will inherit most of the values of the previously created QEMU launcher, etc.

jonahgraham, 2023-06-09:
I'm going to experiment if I can make something without a checkbox as @TommyMurphyTM1234 suggests that also explicitly shows what is going on and defaults as expected.

ilg-ul, 2023-06-09:
Looks good to me. I'll give it a try on Monday, this weekend I'll not be at my computer.

jonahgraham, 2023-06-19:
@MicBiso - this change has now been merged and a development build has been created by the CI. Can you test the new feature using the p2 site https://download.eclipse.org/embed-cdt/builds/develop/p2/?

MicBiso, 2023-06-20:
@jonahgraham - thanks.
I'll give it a go asap.

MicBiso, 2023-06-20:
It works as expected:

[image]

Thanks again.

ilg-ul, 2023-06-20:
Great!

Thank you Jonah for your contribution.

If @TommyMurphyTM1234 has no further suggestions (for this and for #566), I'll prepare a new release in the next few days.

PR #571 (2023-06-19): Add ability to customize GDB port when auto-starting gdb server
When starting a gdbserver (like openocd) connected to a multi-core SoC the gdb server may open multiple ports. This change allows user to configure which port GDB connect to rather than forcing GDB to connect to the first port.

When the ports do not match a warning triangle is displayed which can be clicked on to restore the matching values.

When a user changes the "GDB port" in the "Start TYPE locally" section that is reflected in the "Port number" in the "Remote Target" section.

Changes to "Port number" in the "Remote Target" [... cut]
commit 413c0a39 (2023-06-19): Add ability to customize GDB port when auto-starting gdb server (#571)

When starting a gdbserver (like openocd) connected to a multi-core
SoC the gdb server may open multiple ports. This change allows
user to configure which port GDB connect to rather than forcing
GDB to connect to the first port.

When the ports do not match a warning triangle is displayed which
can be clicked on to restore the matching values.

When a user changes the "GDB port" in the "Start TYPE locally"
section that is reflected in the "Port number" in the "Remote Target"
section.

Changes to "Port number" in the "Remote [... cut]