eclipse-embed-cdt/eclipse-plugins #572: openocd plugin has dependency on managedPlugin which is not given in the manifest file

jantje, 2023-06-11:
The openocd uses the propertytester uses the property org.eclipse.embedcdt.managedbuild.cross.core.isGnuMcu as you can see below
https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/752ae80fe82e14770728ebd18626de9d183ca625/plugins/org.eclipse.embedcdt.debug.gdbjtag.openocd.ui/plugin.xml#L60
The property is defined in the plugin.xml of the managedBuild plugin
https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/752ae80fe82e14770728ebd18626de9d183ca625/plugins/org.eclipse.embedcdt.managedbuild.cross.core/plugin.xml#LL1369C1-L1377C14
However the openocd is not marked as dependent on managedbuild (and there is no reason it should be)

People who only install the openocd plugin to enable openocd debugging will get an stack trace each time they right clik a project in the project explorer (as described here https://github.com/eclipse-cdt/cdt/discussions/415)

```
!ENTRY org.eclipse.core.expressions 4 201 2023-06-11 22:34:10.404
!MESSAGE No property tester contributes a property org.eclipse.embedcdt.managedbuild.cross.core.isGnuMcu to type class org.eclipse.core.internal.resources.Project
!STACK 1
org.eclipse.core.runtime.CoreException: No property tester contributes a property org.eclipse.embedcdt.managedbuild.cross.core.isGnuMcu to type class org.eclipse.core.internal.resources.Project
	at org.eclipse.core.internal.expressions.TypeExtensionManager.getProperty(TypeExtensionManager.java:131)
	at org.eclipse.core.expressions.TestExpression.evaluate(TestExpression.java:104)
	at org.eclipse.core.expressions.CompositeExpression.evaluateAnd(CompositeExpression.java:54)
	at org.eclipse.core.expressions.AndExpression.evaluate(AndExpression.java:36)
	at org.eclipse.ui.internal.dialogs.RegistryPageContributor.failsEnablement(RegistryPageContributor.java:284)
	at org.eclipse.ui.internal.dialogs.RegistryPageContributor.isApplicableTo(RegistryPageContributor.java:227)
	at org.eclipse.ui.internal.dialogs.PropertyPageContributorManager.getApplicableContributors(PropertyPageContributorManager.java:228)
	at org.eclipse.ui.internal.dialogs.PropertyPageContributorManager.getApplicableContributors(PropertyPageContributorManager.java:246)
	at org.eclipse.ui.dialogs.PropertyDialogAction.hasPropertyPagesFor(PropertyDialogAction.java:106)
	at org.eclipse.ui.dialogs.PropertyDialogAction.isApplicableForSelection(PropertyDialogAction.java:147)
	at org.eclipse.ui.dialogs.PropertyDialogAction.isApplicableForSelection(PropertyDialogAction.java:128)
	at org.eclipse.jdt.ui.actions.OpenViewActionGroup.fillContextMenu(OpenViewActionGroup.java:275)
	at org.eclipse.jdt.ui.actions.NavigateActionGroup.fillContextMenu(NavigateActionGroup.java:102)
	at org.eclipse.jdt.internal.ui.actions.CompositeActionGroup.fillContextMenu(CompositeActionGroup.java:78)
	at org.eclipse.jdt.internal.ui.packageview.PackageExplorerActionGroup.fillContextMenu(PackageExplorerActionGroup.java:281)
	at org.eclipse.jdt.internal.ui.packageview.PackageExplorerPart.menuAboutToShow(PackageExplorerPart.java:787)
	at org.eclipse.jface.action.MenuManager.fireAboutToShow(MenuManager.java:338)
	at org.eclipse.jface.action.MenuManager.handleAboutToShow(MenuManager.java:468)
	at org.eclipse.jface.action.MenuManager$2.menuShown(MenuManager.java:495)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:259)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4256)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1066)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1090)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1071)
	at org.eclipse.swt.widgets.Control.WM_INITMENUPOPUP(Control.java:5140)
	at org.eclipse.swt.widgets.Control.windowProc(Control.java:4772)
	at org.eclipse.swt.widgets.Canvas.windowProc(Canvas.java:340)
	at org.eclipse.swt.widgets.Decorations.windowProc(Decorations.java:1478)
	at org.eclipse.swt.widgets.Shell.windowProc(Shell.java:2284)
	at org.eclipse.swt.widgets.Display.windowProc(Display.java:5022)
	at org.eclipse.swt.internal.win32.OS.TrackPopupMenu(Native Method)
	at org.eclipse.swt.widgets.Menu._setVisible(Menu.java:237)
	at org.eclipse.swt.widgets.Display.runPopups(Display.java:4095)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3636)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1155)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1046)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:643)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:550)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:171)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.pde.internal.junit.runtime.NonUIThreadTestApplication.runApp(NonUIThreadTestApplication.java:53)
	at org.eclipse.pde.internal.junit.runtime.NonUIThreadTestApplication.start(NonUIThreadTestApplication.java:48)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:136)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:402)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:596)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1467)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1440)
!SUBENTRY 1 org.eclipse.core.expressions 4 201 2023-06-11 22:34:10.404
!MESSAGE No property tester contributes a property org.eclipse.embedcdt.managedbuild.cross.core.isGnuMcu to type class org.eclipse.core.internal.resources.Project
```

### Steps to Reproduce

1. add openOCD plugin to a newly installed cdt environment
2. create a managed build project
3. right click the project in project properties

I would expect that this keeps the console clean.
The console however dumps the stacktrace given above 2 times.

ruspl-afed, 2023-06-12:
@ilg-ul do you think `org.eclipse.embedcdt.core` bundle could be good new home for `GnuMcuPropertyTester` to fix the reported problem?

ilg-ul, 2023-06-12:
You mean to relocate the definition of `isGnuMcu`

```
	<extension point="org.eclipse.core.expressions.propertyTesters">
		<propertyTester
			class="org.eclipse.embedcdt.managedbuild.cross.core.GnuMcuPropertyTester"
			id="org.eclipse.embedcdt.managedbuild.cross.core.propertyTester"
			namespace="org.eclipse.embedcdt.managedbuild.cross.core"
			properties="isGnuMcu"
			type="java.lang.Object">
		</propertyTester>
	</extension>
```

from `org.eclipse.embedcdt.managedbuild.cross.core` to `org.eclipse.embedcdt.core`?

I guess we can do this. Are there any compatibility issues?

jantje, 2023-06-12:
My 2 cent
It is probably lots easier to use/create a GNU MCU projects nature (which is dependent on the org.eclipse.cdt.core.cnature)
and filter on the GNU MCU projects nature instead of org.eclipse.cdt.core.cnature and remove the enabled when

ruspl-afed, 2023-06-12:
> It is probably lots easier to use/create a GNU MCU projects nature (which is dependent on the org.eclipse.cdt.core.cnature)

It would be easier in the beginning, now it is harder, since people already have a lot of existing projects that don't have this nature and adding a nature now will add even more problems to solve.

jantje, 2023-06-12:
>It would be easier in the beginning, now it is harder, since people already have a lot of existing projects that don't have this nature and adding a nature now will add even more problems to solve.

I'm not fully agreeing here. Adding a nature is not that hard and as you can verify whether a project is a gnu mcu project you can create a listener and add the nature at project open if it is missing.

jonahgraham, 2023-06-12:
> Adding a nature is not that hard

IIUC the issue is not adding a nature for new projects, but that we still need the other version as all user's existing projects won't have the nature. Is that what you were referring to @ruspl-afed?

ilg-ul, 2023-06-27:
Any suggestion on how to proceed with this issue? 

Would adding a new tester like `isEmbeddedCdt` to the core plug-in and leaving the current tester as deprecated be acceptable?

jantje, 2023-06-28:
>Would adding a new tester like isEmbeddedCdt to the core plug-in and leaving the current tester as deprecated be acceptable?

If openocd plugin would reference isEmbeddedCdt tester; to me that would be fine as IMHO openocd plugin has a dependency on core plugin in the manifest file.

ilg-ul, 2023-07-24:
Fixed on 2023-07-24.

closed 2023-07-24 by ilg-ul