eclipse-embed-cdt/eclipse-plugins #579: Feature request: add live watch support when debug in eclipse ide

fanghuaqi, 2023-06-28:
Hello, @ilg-ul , I am using this eclipse plugins to debug riscv projects, and we found current debug don't support live watch on register and expression, and riscv support access memory via system bus, maybe when program running we can access the register and memory via system bus.

[image]

similar issue have been discussed here https://www.eclipse.org/forums/index.php/t/440474/

and similar work have been done in arm eclipse ide, see https://mcuoneclipse.com/2013/01/27/live-view-for-variables-and-memory/

I am wondering whether we can add this feature in this plugin? If possible, any idea in how to implement it, maybe when we have resource, we can try to work on it.

Thanks
Huaqi

ilg-ul, 2023-06-28:
> support live watch  ...   whether we can add this feature?

Well, this is by all means not a quick patch but a complex project, and I don't even know which JTAG probes would support it. 

> ... maybe when we have resource ...

If someone decides to dedicate resources to implement it, sure, we can consider it.

TommyMurphyTM1234, 2023-06-28:
> and we found current debug don't support live watch on register and expression, and riscv support access memory via system bus

Isn't it a prerequisite here that something like OpenOCD supports these features before the plugins can leverage them? Does it support them?

fanghuaqi, 2023-06-28:
> Well, this is by all means not a quick patch but a complex project, and I don't even know which JTAG probes would support it.

Hi @ilg-ul , this is a feature of riscv debug spec, as I know openocd[1], jlink [2] support it.

> Isn't it a prerequisite here that something like OpenOCD supports these features before the plugins can leverage/orchestrate them? Does it support them already? If not then adding support there would be a better place to start?

Hi @TommyMurphyTM1234 , openocd support it via config in openocd cfg file like `riscv set_mem_access sysbus`, but when we debug program in eclipse, and click run it, then you will not be able to view the expression value and memory content, other ide allow periodly monitor these values by a configured time such as 0.5s, but I didn't know how we can achive it in current plugin.

--

[1]: https://openocd.org/doc/html/Architecture-and-Core-Commands.html#RISC_002dV-Debug-Configuration-Commands
[2]: https://wiki.segger.com/J-Link_RISC-V

TommyMurphyTM1234, 2023-06-28:
> Hi @ilg-ul , this is a feature of riscv debug spec, as I know openocd[1](https://openocd.org/doc/html/Architecture-and-Core-Commands.html#RISC_002dV-Debug-Configuration-Commands), jlink [2](https://wiki.segger.com/J-Link_RISC-V) support it.

Can you clarify where access to registers is provided by the current OpenOCD imlementation please?

> Hi @TommyMurphyTM1234 , openocd support it via config in openocd cfg file like `riscv set_mem_access sysbus`, but when we debug program in eclipse, and click run it, then you will not be able to [...]

Does GDB via the command line console in Eclipse allow you to exercise these features?

fanghuaqi, 2023-06-28:
> Can you clarify where access to registers is provided by the current OpenOCD imlementation please?

This feature seems not available in openocd, hardware may not support it.

fanghuaqi, 2023-06-28:
> Does GDB via the command line console in Eclipse - or eben GDB standalone outside of Eclipse altogether - allow you to exercise these features?

No, I tried send gdb command in eclipse debugger console when target is running, not response is back.

TommyMurphyTM1234, 2023-06-28:
> > Does GDB via the command line console in Eclipse - or eben GDB standalone outside of Eclipse altogether - allow you to exercise these features?
> 
> No, I tried send gdb command in eclipse debugger console when target is running, not response is back.

@ilg-ul can correct me if I'm wrong but I believe that the plugins will only be able to exercise features that GDB allows to be used in the first place. 

Sounds to me like this is potentially a mini project that needs to consider some or all of the following - and maybe others - and not just the plugins:

1. RISC-V debug spec and debug capabilities - in particular what target resources are accessible via SBA while the target is free running
2. OpenOCD (and other GDB servers) support for these capabilities
3. GDB (and LLDB?) support for same
4. What capabilities other commercial/non open-source tools provide
5. Eclipse plugin support for these capabilities

ilg-ul, 2023-06-28:
> I believe that the plugins will only be able to exercise features that GDB allows to be used in the first place.

That's correct.

Only if you can use a feature in a terminal we should consider changes to the debug plug-ins.

fanghuaqi, 2023-07-04:
I collect some live watch features supported in other IDE for riscv:

see https://wwwfiles.iar.com/riscv/EWRISCV_DebuggingGuide.pdf

[image]

ilg-ul, 2023-07-04:
Just that, as far as I know, IAR does not use GDB. Neither SEGGER.

TommyMurphyTM1234, 2023-07-04:
Actually, this might be of relevance:

* https://visualgdb.com/documentation/livevars/

If VisualGDB can leverage GDB and OpenOCD to provide this feature then perhaps the plugins can too?