eclipse-embed-cdt/eclipse-plugins #581: GDB fails when workspace path has extended ASCII characters

silabs-daperez, 2023-07-11:
### Description
* Hello. When a workspace path or folder name has an extended ASCII character within it (e.g. é - e with accent) , a GDB session fails to launch. 
* One of our customers has reported this problem and we've reproduced it, though the exception stacktraces are different. 
* The workaround is to use a workspace path with only ASCII characters.
* Thank you for the help!

### Steps to Reproduce

1. Create workspace folder with extended ASCII character in the folder name, e.g. étudiant.
2. Launch Eclipse and switch to  workspace folder with extended ASCII character in the folder name.
3. Create project for device that uses Segger JLink GDB server
4. Launch GDB session for project with device
5. Confirm GDB session fails to launch

**Expected behaviour:** GDB session launches successfully and user can debug application.

**Actual behaviour:** GDB session fails to launch with exceptions thrown. Please see the exceptions at bottom of this post.

### Versions

* Plug-ins version (Target Platform): CDT 10.5.0 / Embed CDT 6.1.2 (Note: J-Link plugins are used primarily)
* Eclipse version: 4.28
* Java version: JDK 17.0.6 Amazon Correto / Plug-ins execute with JavaSE-11
* Operating system: Windows 10 (language: English US)
* Toolchain version: ARM GNU GCC v10.3.1

### Exceptions

**Exception encountered by me**:

> !ENTRY org.eclipse.cdt.dsf.gdb 4 5012 2023-07-07 17:05:38.440
!MESSAGE Error in final launch sequence
!STACK 1
org.eclipse.core.runtime.CoreException: Failed to execute MI command:
-environment-cd C:/SiliconLabs/workspaces/étudiant_Test/project_name
Error message from debugger back end:
C:/SiliconLabs/workspaces/étudiant_Test/project_name: No such file or directory.
	at org.eclipse.cdt.dsf.concurrent.Query.get(Query.java:112)
	at org.eclipse.embedcdt.debug.gdbjtag.jlink.core.dsf.LaunchConfigurationDelegate.launchDebugSession(LaunchConfigurationDelegate.java:460)
	at org.eclipse.embedcdt.debug.gdbjtag.jlink.core.dsf.LaunchConfigurationDelegate.launchDebugger(LaunchConfigurationDelegate.java:184)
	at org.eclipse.embedcdt.debug.gdbjtag.jlink.core.dsf.LaunchConfigurationDelegate.launch(LaunchConfigurationDelegate.java:157)
	at com.silabs.ss.tool.ide.arm.api.debug.gdb.StudioGdbLaunchConfigurationDelegate.launch(StudioGdbLaunchConfigurationDelegate.java:114)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:803)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:716)
	at org.eclipse.debug.internal.ui.DebugUIPlugin.buildAndLaunch(DebugUIPlugin.java:1021)
	at org.eclipse.debug.internal.ui.DebugUIPlugin$2.run(DebugUIPlugin.java:1224)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
Caused by: java.lang.Exception: C:/SiliconLabs/workspaces/étudiant_Test/project_name: No such file or directory.
	at org.eclipse.cdt.dsf.mi.service.command.AbstractMIControl$RxThread.processMIOutput(AbstractMIControl.java:1013)
	at org.eclipse.cdt.dsf.mi.service.command.AbstractMIControl$RxThread.run(AbstractMIControl.java:842)
!SUBENTRY 1 org.eclipse.cdt.dsf.gdb 4 10004 2023-07-07 17:05:38.449
!MESSAGE Failed to execute MI command:
-environment-cd C:/SiliconLabs/workspaces/étudiant_Test/project_name
Error message from debugger back end:
C:/SiliconLabs/workspaces/étudiant_Test/project_name: No such file or directory.
!STACK 0
java.lang.Exception: C:/SiliconLabs/workspaces/étudiant_Test/project_name: No such file or directory.
	at org.eclipse.cdt.dsf.mi.service.command.AbstractMIControl$RxThread.processMIOutput(AbstractMIControl.java:1013)
	at org.eclipse.cdt.dsf.mi.service.command.AbstractMIControl$RxThread.run(AbstractMIControl.java:842)
>

**Exception reported by customer**:

>!ENTRY org.eclipse.cdt.dsf.gdb 4 104 2023-02-14 17:04:53.252
!MESSAGE Program file does not exist
!STACK 0
java.io.FileNotFoundException: C:\Users\étudiant\SimplicityStudio\v5_workspace\project_name\GNU ARM v10.3.1 - Default\project_name.axf not found
    at org.eclipse.cdt.dsf.gdb.launching.LaunchUtils.verifyProgramPath(LaunchUtils.java:130)
    at org.eclipse.cdt.dsf.gdb.launching.GdbLaunchDelegate.checkBinaryDetails(GdbLaunchDelegate.java:330)
    at org.eclipse.embedcdt.debug.gdbjtag.jlink.core.dsf.LaunchConfigurationDelegate.checkBinaryDetails(LaunchConfigurationDelegate.java:511)
    at com.silabs.ss.tool.ide.arm.api.debug.gdb.StudioGdbLaunchConfigurationDelegate.checkBinaryDetails(StudioGdbLaunchConfigurationDelegate.java:610)
    at org.eclipse.embedcdt.debug.gdbjtag.jlink.core.dsf.LaunchConfigurationDelegate.launchDebugSession(LaunchConfigurationDelegate.java:237)
    at org.eclipse.embedcdt.debug.gdbjtag.jlink.core.dsf.LaunchConfigurationDelegate.launchDebugger(LaunchConfigurationDelegate.java:184)
    at org.eclipse.embedcdt.debug.gdbjtag.jlink.core.dsf.LaunchConfigurationDelegate.launch(LaunchConfigurationDelegate.java:157)
    at com.silabs.ss.tool.ide.arm.api.debug.gdb.StudioGdbLaunchConfigurationDelegate.launch(StudioGdbLaunchConfigurationDelegate.java:114)
    at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:803)
    at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:716)
    at org.eclipse.debug.internal.ui.DebugUIPlugin.buildAndLaunch(DebugUIPlugin.java:1021)
    at org.eclipse.debug.internal.ui.DebugUIPlugin$2.run(DebugUIPlugin.java:1224)
    at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
>

jonahgraham, 2023-07-11:
This bug, while showing up while you are using embed-cdt is almost certainly from the partner project [Eclipse CDT](https://github.com/eclipse-cdt/cdt). 

Can you supply the [gdb traces](https://github.com/eclipse-cdt/cdt/tree/main/FAQ#ive-been-asked-for-gdb-traces-where-can-i-find-them) of the issue please? 

Coincidentally I have just been looking at a [similar issue](https://github.com/eclipse-cdt-cloud/cdt-gdb-adapter/issues/275) and see that CDT has numerous places where, when communicating with GDB, there are encoding differences that are not properly accounted for. For example, even if you manage to launch you are likely to hit other problems such as setting breakpoints in such files.

> The workaround is to use a workspace path with only ASCII characters.

It is likely that this is the workaround that will be needed for a while. However we can get a proper solution faster if you can contribute some PRs?

The other mitigation that is employed by some vendors is to prevent creation of projects with non-ASCII characters in the first place.

silabs-daperez, 2023-07-11:
Thank you, Jonah!
### Questions & Answers
> Can you supply the [gdb traces](https://github.com/eclipse-cdt/cdt/tree/main/FAQ#ive-been-asked-for-gdb-traces-where-can-i-find-them) of the issue please?

Please see the GDB traces at the bottom of this post.

> For example, even if you manage to launch you are likely to hit other problems such as setting breakpoints in such files.

Yes, we've encountered such scenarios as well. One that we're reproduced is after GDB launches successfully, it cannot display source code because the SourceLocator failed with workspace path that has extended ASCII characters.

>However we can get a proper solution faster if you can contribute some PRs?

I can try. Where are the instructions for applying for CDT committer status? Does CDT committer status extend to Eclipse Embed CDT, or are they separate?

### GDB Traces
Here's the trace for my failed GDB session launch.

>135,414 2-gdb-show language
135,418 2^done,value="auto"
135,418 (gdb) 
135,419 3-data-evaluate-expression "sizeof (void*)"
135,433 3^done,value="4"
135,433 (gdb) 
135,434 4-gdb-set language auto
135,449 4^done
135,449 (gdb) 
135,450 5-interpreter-exec console "show endian"
135,465 ~"The target endianness is set automatically (currently little endian).\n"
135,465 5^done
135,465 (gdb) 
135,467 6-gdb-version
135,481 ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"
135,481 ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
135,481 ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is fre\
e software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitt\
ed by law."
135,481 ~"\nType \"show copying\" and \"show warranty\" for details.\n"
135,481 ~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\n"
135,481 ~"Type \"show configuration\" for configuration details.\n"
135,481 ~"For bug reporting instructions, please see:\n"
135,481 ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
135,481 ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/\
software/gdb/documentation/>."
135,481 ~"\n\n"
135,482 ~"For help, type \"help\".\n"
135,482 ~"Type \"apropos word\" to search for commands related to \"word\".\n"
135,482 6^done
135,482 (gdb) 
135,484 7-environment-cd C:/SiliconLabs/workspaces/étudiant_Test/project_name
135,497 7^error,msg="C:/SiliconLabs/workspaces/\303\251tudiant_Test/project_name: No such file or d\
irectory."
135,498 (gdb) 
135,499 8-gdb-exit
135,513 8^exit
>

TommyMurphyTM1234, 2023-07-11:
Does GDB work standalone (in a terminal without any associated Eclipse/CDT/Embedded CDT) when using such paths/filenames containing accented characters? Ditto for OpenOCD assuming that you're using it?

jonahgraham, 2023-07-11:
> I can try. Where are the instructions for applying for CDT committer status? 

No special status is needed to contribute a Pull Request - only that you have a valid [ECA on file](https://github.com/eclipse-cdt/cdt/blob/main/CONTRIBUTING.md#eclipse-contributor-agreement) with the Eclipse Foundation. 

> 135,484 7-environment-cd C:/SiliconLabs/workspaces/étudiant_Test/project_name
> 135,497 7^error,msg="C:/SiliconLabs/workspaces/\303\251tudiant_Test/project_name: No such file or d
irectory."

Looks like encoding of the `é` is inconsistent and therefore it comes out wrong. This is the same issue as https://github.com/microsoft/vscode-cpptools/issues/1998 for VSCode and there the workaround was to ensure Windows is set to UTF-8 - https://github.com/microsoft/vscode-cpptools/issues/1998#issuecomment-1315933181 - interestingly vscode-cpptools have closed the issue pointing back at fixes needed in GDB. @TommyMurphyTM1234's question is quite relevant here, if you cannot do the `-environment-cd` when running gdb outside of Eclipse (in MI mode), then there isn't anything we can do in CDT.

silabs-daperez, 2023-07-12:
* Thanks for the feedback so far. I see your point.
* The GDB client we are using is version `10.2.90.20210621-git`, which we obtained from `GNU Arm Embedded Toolchain 10.3-2021.10`.
* I tried your suggestion of using GDB outside of Eclipse. I launched the GDB client on command line with arguments `--interpreter=mi --nx`. Then I ran the `file <path_to_program>` command to load the program from command line, but it fails to handle the path with extended ASCII characters, resulting in error `No such file or directory.`.  See command line output further below.
* I'll try upgrading the GDB client to a newer version to see if that resolves the issue.

> $ arm-none-eabi-gdb.exe --interpreter=mi --nx
arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
=thread-group-added,id="i1"
~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"
~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
~"\nType \"show copying\" and \"show warranty\" for details.\n"
~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\n"
~"Type \"show configuration\" for configuration details.\n"
~"For bug reporting instructions, please see:\n"
~"<https://www.gnu.org/software/gdb/bugs/>.\n"
~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
~"\n\n"
~"For help, type \"help\".\n"
~"Type \"apropos word\" to search for commands related to \"word\".\n"
(gdb)
file C:/SiliconLabs/workspaces/étudiant_Test/project_name/GNU\ ARM\ v10.3.1\ -\ Default/project_name.axf
&"file C:/SiliconLabs/workspaces/\202tudiant_Test/project_name/GNU\\ ARM\\ v10.3.1\\ -\\ Default/project_name.axf\n"
&"C:/SiliconLabs/workspaces/\202tudiant_Test/project_name/GNU ARM v10.3.1 - Default/project_name.axf: No such file or directory.\n"
^error,msg="C:/SiliconLabs/workspaces/\202tudiant_Test/project_name/GNU ARM v10.3.1 - Default/project_name.axf: No such file or directory."
(gdb)
>

TommyMurphyTM1234, 2023-07-12:
> * The GDB client we are using is version `10.2.90.20210621-git`, which we obtained from `GNU Arm Embedded Toolchain 10.3-2021.10`.

GNU ARM Embedded seems to have stopped at 10.2

* https://developer.arm.com/downloads/-/gnu-rm

 and been superseded by the Arm GNU Toolchain:

* https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

You could also try @ilg-ul's xPack toolchain:

* https://xpack.github.io/dev-tools/arm-none-eabi-gcc/releases/

> * I tried your suggestion of using GDB outside of Eclipse. I launched the GDB client on command line with arguments `--interpreter=mi --nx`. Then I ran the `file <path_to_program>` command to load [...]

So it sounds like GDB itself also (?) has problems handling "extended" characters?
So there may be several layers of tools that need to be updated to deal with this which is probably not a trivial task.

silabs-daperez, 2023-07-19:
### Status Update
* Sorry to take so long to reply. I got sidetracked with other tasks.
* I also tried the latest GDB client `arm-none-eabi-gdb.exe` from [arm-gnu-toolchain-12.2.rel1-mingw-w64-i686-arm-none-eabi](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads) and get the same error `No such file or directory.`. Below is the the console output.
* @TommyMurphyTM1234 Thanks for the suggestion to use @ilg-ul's xPack toolchain. I get the same error using GDB client `arm-none-eabi-gdb.exe` within `xpack-arm-none-eabi-gcc-12.2.1-1.2-win32-x64`. Below is the console output.
* At this point, we'll have to rely on the workarounds (1. use paths with only ASCII chars 2. [Enable Beta: Use Unicode UTF-8 for worldwide language support](https://github.com/microsoft/vscode-cpptools/issues/1998)  )  until this is fixed in the GDB client.
* Thanks for the help, @jonahgraham and @TommyMurphyTM1234 !

### Console Output: Arm GNU Toolchain 12.2.Rel1
> $arm-none-eabi-gdb.exe --interpreter=mi --nx
=thread-group-added,id="i1"
~"GNU gdb (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24)) 12.1.90.20221210-git\n"
~"Copyright (C) 2022 Free Software Foundation, Inc.\n"
~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
~"\nType \"show copying\" and \"show warranty\" for details.\n"
~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\n"
~"Type \"show configuration\" for configuration details.\n"
~"For bug reporting instructions, please see:\n"
~"<https://bugs.linaro.org/>.\n"
~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
~"\n\n"
~"For help, type \"help\".\n"
~"Type \"apropos word\" to search for commands related to \"word\".\n"
(gdb)
file C:/SiliconLabs/workspaces/étudiant_Test/mg22_Z3Switch/GNU\ ARM\ v10.3.1\ -\ Default/mg22_Z3Switch.axf
&"file C:/SiliconLabs/workspaces/\202tudiant_Test/mg22_Z3Switch/GNU\\ ARM\\ v10.3.1\\ -\\ Default/mg22_Z3Switch.axf\n"
&"C:/SiliconLabs/workspaces/\202tudiant_Test/mg22_Z3Switch/GNU ARM v10.3.1 - Default/mg22_Z3Switch.axf: No such file or directory.\n"
^error,msg="C:/SiliconLabs/workspaces/\202tudiant_Test/mg22_Z3Switch/GNU ARM v10.3.1 - Default/mg22_Z3Switch.axf: No such file or directory."
(gdb)
quit
&"quit\n"
>

### Console Output: xpack-arm-none-eabi-gcc-12.2.1-1.2-win32-x64
> $arm-none-eabi-gdb.exe --interpreter=mi --nx
=thread-group-added,id="i1"
~"GNU gdb (xPack GNU Arm Embedded GCC x86_64) 12.1.90.20221210-git\n"
~"Copyright (C) 2022 Free Software Foundation, Inc.\n"
~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
~"\nType \"show copying\" and \"show warranty\" for details.\n"
~"This GDB was configured as \"--host=x86_64-w64-mingw32 --target=arm-none-eabi\".\n"
~"Type \"show configuration\" for configuration details.\n"
~"For bug reporting instructions, please see:\n"
~"<https://www.gnu.org/software/gdb/bugs/>.\n"
~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
~"\n\n"
~"For help, type \"help\".\n"
~"Type \"apropos word\" to search for commands related to \"word\".\n"
(gdb)
file C:/SiliconLabs/workspaces/étudiant_Test/mg22_Z3Switch/GNU\ ARM\ v10.3.1\ -\ Default/mg22_Z3Switch.axf
&"file C:/SiliconLabs/workspaces/\202tudiant_Test/mg22_Z3Switch/GNU\\ ARM\\ v10.3.1\\ -\\ Default/mg22_Z3Switch.axf\n"
&"C:/SiliconLabs/workspaces/\202tudiant_Test/mg22_Z3Switch/GNU ARM v10.3.1 - Default/mg22_Z3Switch.axf: No such file or directory.\n"
^error,msg="C:/SiliconLabs/workspaces/\202tudiant_Test/mg22_Z3Switch/GNU ARM v10.3.1 - Default/mg22_Z3Switch.axf: No such file or directory."
(gdb)
>

jonahgraham, 2023-07-19:
Thanks @silabs-daperez for coming back to this. I am going to close this issue as the bug is in GDB. I am glad you have a workaround.

closed 2023-07-19 by jonahgraham

Gadicuz, 2024-04-28:
I have the same problem and I'd like **to propose a solution**.

<details>
<summary>TL;DR</summary>
<pre>
 - use system local encoding (win1251, win1252 and so on) to encode/decode command exchange with gdb
 - required encoding could be optional and might be configured in launch.json
</pre>
</details>

Prerequisites:
- platform = win32
- non-ASCII character(s) in a file path (extension path or project path)

Fact: **gdb uses system default encoding for all scripting stuff**
_This is win1251 in my case._

Let's consider the following code:

```js
var spawn = require('child_process').spawn;
var iconv = require('iconv-lite');

var encoding = 'win1251'; // specify your windows default encoding (gdb uses it for scripting)
var dcd = (data) => iconv.decode(data, encoding);  // transcode incoming data
var raw = (data) => data.toString();  // use raw() to test skipping transcoding incoming data

var gdb = spawn("gdb.exe");
gdb.stdout.on('data', (data) => console.log(dcd(data)));
gdb.stderr.on('data', (data) => console.error(dcd(data)));

var file0 = "file0-ascii.txt";     // ASCII-only filename (exists)
var file1 = "file1-ascii.txt";     // ASCII-only filename (exists)
var file2 = "file2-неаски.txt";    // non-ASCII filename (exists)
var file3 = "file3-нетути.txt";    // non-ASCII filename (doesn't exist)

// use utf-8 encoding
gdb.stdin.write(`interpreter-exec console "source ${file0}"\n`);  // [A]
gdb.stdin.write(`interpreter-exec console "source ${file1}"\n`);  // [B]
gdb.stdin.write(`interpreter-exec console "source ${file2}"\n`);  // [C]
// transcode to gdb-friendly encoding
gdb.stdin.write(iconv.encode(`interpreter-exec console "source ${file2}"\n`, encoding)); // [D]
gdb.stdin.write(iconv.encode(`interpreter-exec console "source ${file3}"\n`, encoding)); // [E]
gdb.stdin.end();
```
Also we have three files (all 'win1251' content encoded, attached below):

<details><summary>_file0-ascii.txt_</summary><pre>
echo <<< Hello ASCII world >>>
</pre></details>

<details><summary>_file1-ascii.txt_</summary><pre>
echo <<< У меня есть не-ASCII друг (I have non-ASCII friend) >>>
source file2-неаски.txt
</pre></details>

<details><summary>_file2-неаски.txt_</summary><pre>
echo <<< Привет не-ASCII мир! >>>
</pre></details>

Running the code we get output:
```
...
For help, type "help".
Type "apropos word" to search for commands related to "word".
[A](gdb) <<< Hello ASCII world >>>
[B](gdb) <<< У меня есть не-ASCII друг (I have non-ASCII friend) >>><<< Привет не-ASCII мир! >>>
[C](gdb)
file2-РЅРµР°СЃРєРё.txt: No such file or directory.
error in command: "source file2-РЅРµР°СЃРєРё.txt".
[D](gdb) <<< Привет не-ASCII мир! >>>
[E](gdb)
file3-нетути.txt: No such file or directory.
error in command: "source file3-нетути.txt".
(gdb) 
```

 - Command [A]: executes simple command file "file0" with ASCII name, no problem at all.
 - Command [B]: executes another command file "file1" with ASCII name. The file outputs some non-ASCII text and executes non-ASCII named file. No problems, everything works!!! If one replaces dcd() with raw() non-ASCII text output will corrupt but all scripts keep to work.

This proves gdb internaly uses local system encoding to work with command files and their contents. Let's run "file2" with interpreter by command, the same file we just ran from within "file1".
 - Command [C]: executes "file2" with non-ASCII name directly and failes. gdb failed to find the file and there is the reason.
<details>
  <summary>Explanation</summary>
<pre>
 non-ASCII part of file2 string is      "\u{043D}\u{0435}\u{0430}\u{0441}\u{043A}\u{0438}"
 utf-8 encoding is                      "\xD0\xBD\xD0\xB5\xD0\xB0\xD1\x81\xD0\xBA\xD04\xB8" 
 written to stdin as utf-8 encoding     "\xD0\xBD\xD0\xB5\xD0\xB0\xD1\x81\xD0\xBA\xD04\xB8" (unchanged)
        gdb uses system default encoding so every byte is a character, hence failed to locate file
 returned in error message as           "\xD0\xBD\xD0\xB5\xD0\xB0\xD1\x81\xD0\xBA\xD04\xB8" (unchanged)
        by default input stream is decoded to string using utf-8, so filename with raw() transcoding seems ok, BUT...
        ... the real encoding from gdb (including this error message) is system default, so we need to
 transcode name from win1251 bytes      "  Р   Ѕ   Р   µ   Р   °   С   Ѓ   Р   є   Р   ё"  (win1251 decoded)
      This is **the** filename gdb tried to find due to encoding mismatch,
      it also does interpret every byte as a filename character so 
      names "file2-неаски.txt" and "file2-РЅРµР°СЃРєРё.txt" are not the same.
</pre>
</details>

Let's encode "file2" name with win1251 and try again.
 - Command [D]: same as command [C] but filename is encoded with local encoding (gdb-friendly). Works perfectly!
 - Command [E]: encoded filename of unexisting file - failed with correct message.

### Conclusion
Use system default encoding to exchange command data with gdb and everything will work!!!

[file0-ascii.txt](https://github.com/eclipse-embed-cdt/eclipse-plugins/files/15142181/file0-ascii.txt)
[file1-ascii.txt](https://github.com/eclipse-embed-cdt/eclipse-plugins/files/15142182/file1-ascii.txt)
[file2-неаски.txt](https://github.com/eclipse-embed-cdt/eclipse-plugins/files/15142183/file2-.txt)