eclipse-embed-cdt/eclipse-plugins #588: Install CMSIS Pack from local .pack file?

deividAlfa, 2023-09-03:
Eclipse version: 2023-06 (4.28.0)
CDT plugin version: 6.4.0
OS: Windows 11 x64

Some vendors (Puya, HSXP) provide plain .pack files, but CMSIS-Pack manager depends on the Keil repo to download the packs.
So far, it seem to be impossible to manually install a .pack file. Is it possible somehow?
I have only found https://github.com/eclipse-embed-cdt/eclipse-plugins/issues/61 and it was never anwered.

TommyMurphyTM1234, 2023-09-03:
Did you try the approach that I sketched out here?

* https://sourceforge.net/p/gnuarmeclipse/feature-requests/98/#c56c

ilg-ul, 2023-09-03:
> CMSIS-Pack manager depends on the Keil repo to download the packs.

Negative. CMSIS-Pack manager can use multiple repos to download packs.

Just create a conforming index and add it to the CMSIS-Pack manager configuration.

> it seem to be impossible to manually install a .pack file

I did not check recently, but the initial CMSIS specs did not define such a use case, so packs can be installed only via an index file.

I did not check, but it might be possible to place the index locally and pass a uri like `file://...`.

But what do you need the CMSIS-Packs for? Apart from the documentation, the plug-ins do not use the .pdsc definitions at all.

For the SVDs, you can directly refer them from the debug plug-ins, you do not need the entire package for this.

deividAlfa, 2023-09-03:
Basically I'd like to have them in the device selection?
Of course, SVD for debugging...

ilg-ul, 2023-09-04:
> have them in the device selection

Yes, this was a nice feature, and initially it made sense assuming the .pdsc files can be used for other purposes too.

But later it proved less suitable, and instead of managing the huge CMSIS packages (some take hundreds of MBs), it looked simpler to just copy the .svd files into the projects and pass them to the debugger directly.

Also the permanent timeouts of the servers hosting the packages and the out-of-sync Keil index were some negative factors that contributed to the CMSIS Packs plug-ins to be no longer actively developed.