eclipse-embed-cdt/eclipse-plugins #592: Feature request: Add an checkbox to enable group linked library

fanghuaqi, 2023-10-17:
Hi @ilg-ul ,

I would like to request a feature like below:

[image]

Adding a checkbox in Settings -> Linker -> Libraries, this checkbox is used to control group libraries or not
- [ ] Group libraries(-Wl,start-group $LIBS -Wl, end-group)

When this checkbox is clicked, the generated Makefile will be like this:

[image]

Otherwise, it will be like this, which is the default behavior

[image]

This feature is used to fix issue like this `If a symbol in that archive is needed to resolve an undefined symbol referred to by an object in an archive that appears later on the command line, the linker would not be able to resolve that reference.`

[image]

See reference links below:

- https://stackoverflow.com/questions/5651869/what-are-the-start-group-and-end-group-command-line-options
- https://ftp.gnu.org/old-gnu/Manuals/ld-2.9.1/html_chapter/ld_2.html

Thanks
Huaqi

ilg-ul, 2023-10-17:
Thank you for the suggestion, I'll think about it.

@jonahgraham, do you know of something similar in CDT? If not, are there any plans to implement it?

fanghuaqi, 2023-10-17:
> Thank you for the suggestion, I'll think about it.
> 
> @jonahgraham, do you know of something similar in CDT? If not, are there any plans to implement it?

Thank you.

TommyMurphyTM1234, 2023-10-17:
In case it's of any relevance, this 10 year old CDT bug relates to the same issue:

* https://bugs.eclipse.org/bugs/show_bug.cgi?id=398355

I'm not sure about "impossible" though, given that there seem to be workarounds - for example:

* https://stackoverflow.com/questions/58817617/how-to-link-intel-mkl-using-eclipse-start-group-end-group

fanghuaqi, 2023-10-17:
Hi @TommyMurphyTM1234 , it seems to be a CDT issue, workaround is not in an clean way, but I will take a try.

TommyMurphyTM1234, 2023-10-17:
> Hi @TommyMurphyTM1234 , it seems to be a CDT issue, workaround is not in an clean way, but I will take a try.

I think it's more or less what @ilg-ul previously suggested for Embedded CDT.
I just posted the links in case they were relevant to the comments about the base CDT.

ilg-ul, 2023-10-17:
> I think it's more or less what @ilg-ul previously suggested for Embedded CDT.

That's correct. With the current implementation there are two solutions:

- add the group as Miscellaneous -> Other linker flags
- add the --start-group/--end-group in the Linker -> Expert settings -> Command line pattern

laomaolaile, 2023-10-17:
Adding to `Linker -> Misc -> Other Objects ` is not working.

[image]

laomaolaile, 2023-10-17:
Put it in other linker flags is also not working

[image]

[image]

laomaolaile, 2023-10-17:
Workaround like this, seems to work, but it is not a good way for normal user.

[image]
[image]
[image]

fanghuaqi, 2024-01-03:
Hi @ilg-ul @jonahgraham @jld01 , I see https://github.com/eclipse-cdt/cdt/issues/608 this issue is fixed, maybe embedded cdt plugin can support this group library feature now using this release https://github.com/eclipse-cdt/cdt/releases/tag/CDT_11_4_0, is there any work to do in this embedded cdt plugin?

laomaolaile, 2024-02-23:
Hi @ilg-ul @jonahgraham @jld01 

I created a MinGW GCC project using eclipse-embedcpp-2023-12-R-win32-x86_64, which already supports the Group libraries feature.

[image]

However, the created risc-v project currently does not support the Group libraries feature.

[image]

If want the risc-v project to also support Group libraries, do need to complete some development work?

If I want to achieve it and contribute, what should I do?

ilg-ul, 2024-02-23:
The ARM & RISC-V managed build plug-ins and the MinGW managed build plug-in are different plug-ins and do not share code, so that button and the associated functionality must be reimplemented.

I currently don't know how complex this can be, the MinGW plug-in must be studied and a similar solution used, if possible.

laomaolaile, 2024-02-26:
Hi @ilg-ul @jld01 

Maybe I expressed wrongly. The key point is not MinGW GCC. I just want to know if  can add Group libraries to the risc-v's GNU IRSC-V Corss C++ Links page, and if it supports this usage after adding it.

[image]

[image]

ilg-ul, 2024-02-26:
> if can add Group libraries to the risc-v's

If we decide to implement it, It must be added to the common plug-in and it'll be inherited to both RISC-V and Arm.

> if it supports this usage after adding it.

I have no experience with this use case.

But you don't have to patch the plug-ins to use it, you can manually define all those options in the Miscellaneous field.

laomaolaile, 2024-03-14:
Hello, this issue has been raised for some time now, and support for it is already available in CDT. May I ask if anyone is currently working on addressing this requirement?

@ilg-ul @jld01

laomaolaile, 2024-03-15:
I have completed the development of the function. How can I share the code?

The demonstration effects in the ARM project are as follows:

When Group libraries unchecked:

[image]

[image]

When Group libraries checked:

[image]

[image]

The demonstration effects in the RISC-V project are as follows:

When Group libraries unchecked:

[image]

[image]

When Group libraries checked:

[image]

[image]

ilg-ul, 2024-03-19:
I took a look at this issue and I'm a bit confused, the solution I saw in CDT did not work in my projects, adding a `LibrariesCommandGenerator` to surround the list of libraries via `-Wl,--start-group %s -Wl,--end-group` failed because the resulting string is expected to have no spaces, otherwise they're escaped with a backslash and the compiler is not happy with them.

My solution was to generate everything as a single `-Wl,` string, like:

```
-Wl,--start-group,-lone,-ltwo,-lthree,--end-group
```

Any comments on this solution? @jld01, @laomaolaile ?

ilg-ul, 2024-03-19:
As window layout, I would add the new option just below the list of libraries, since it applies to them, like this:

[image]

What do you think?

TommyMurphyTM1234, 2024-03-19:
> As window layout, I would add the new option just below the list of libraries, since it applies to them, like this:
> 
> ![Screenshot 2024-03-19 at 21 12 41](https://private-user-images.githubusercontent.com/362732/314198278-77fc0378-ede3-4fa2-930f-ffcc46c7946f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA4NzYyMDEsIm5iZiI6MTcxMDg3NTkwMSwicGF0aCI6Ii8zNjI3MzIvMzE0MTk4Mjc4LTc3ZmMwMzc4LWVkZTMtNGZhMi05MzBmLWZmY2M0NmM3OTQ2Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxOVQxOTE4MjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MjY2ODRlYzYxOTUwMjk3YTVlZDRlODJlZWExY2Y2MGE1MzdlM2JmMDAxNDc0NGI1ODhhNWUxYTY3ZjJkNDY2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.cMIdl9ps56gLR2F9EF69Vx2AGLIWIHf-qL-kDXHTjNA)
> 
> What do you think?

That makes sense to me but unfortunately seems to deviate from where CDT places it, and perhaps all plugins should be consistent?

* https://github.com/eclipse-embed-cdt/eclipse-plugins/issues/592#issuecomment-1875135870

ilg-ul, 2024-03-19:
> unfortunately seems to deviate from where CDT places it

yeah, but the behaviour is also slightly different. and it is not widely used by CDT, probably it is Windows specific, since I don't see it on my Mac.

ilg-ul, 2024-03-19:
@laomaolaile, please test the pre-release from:

- https://download.eclipse.org/embed-cdt/builds/develop/p2/

laomaolaile, 2024-03-20:
I have tested it and it's exactly the function I want. Thank you very much. @ilg-ul

ilg-ul, 2024-03-20:
Thank you for confirming.

I'll proceed with the 6.6.0 release.

closed 2024-04-03 by ilg-ul