eclipse-embed-cdt/eclipse-plugins #597: make: unknown option -- jobserver-auth=gmake_semaphore_172 when using post-build steps

fanghuaqi, 2023-11-13:
Hi @ilg-ul ,

When I am using `C/C++ Build -> Settings -> Build Steps` and add post build steps as below, and save it, and then rebuild it, I will face this issue in Windows.

[image]

[image]

I think this is caused by `&&` not recognized by windows, I need to disable auto makefile generation and change it like this, then this issue will disappear.

[image]

If I want to fix it in this plugin, how can I fix it.

Thanks

fanghuaqi, 2023-11-13:
Hi, we tested that, if we change back the windows build tools to 4.3.0-1, not using 4.4.0 or later, this issue will be solved, as I can see, it seems, make version changed from 4.3 to 4.4.

Thanks

ilg-ul, 2023-11-13:
According to some reports, make 4.4 may have problems. As a workaround, use 4.3.

fanghuaqi, 2023-11-13:
Okay, thanks. Will use 4.3

fanghuaqi, 2023-11-13:
Sorry @ilg-ul , it is not caused by make version change, it seems caused by the busybox utils version update . I used make version 4.4, and busybox 4.3.0, it works.

ilg-ul, 2023-11-13:
For completeness, could you also test [v4.4.0-1](https://github.com/xpack-dev-tools/windows-build-tools-xpack/releases/tag/v4.4.0-1)?

fanghuaqi, 2023-11-14:
Same issue happened. @ilg-ul 

[image]

I even tried with latest https://frippery.org/busybox/ , same issue happened. I think maybe busybox changed something

~~~
BusyBox v1.37.0-FRP-5181-g5c1a3b00e (2023-08-18 12:14:59 BST)
~~~

Also tried with old releases of busybox, tested https://frippery.org/files/busybox/ that 	`busybox-w32-FRP-4716-g31467ddfc.exe` is the last workable version.

This is the first not workable version https://frippery.org/busybox/release-notes/FRP-4784.html changelog.

Thanks

ilg-ul, 2023-11-14:
Thank you!

I suggest you report this to upstream busybox.

rmyorston, 2023-11-14:
Since FRP-4784 busybox-w32 has included a `make` applet. From the errors reported it looks as though this is being used when GNU make was intended.

It's possible to override the busybox-w32 shell's preference for the internal `make` by setting an environment variable: `BB_OVERRIDE_APPLETS=make`.

ilg-ul, 2023-11-14:
Thank you, Ron.

Is there any configuration option to disable the applet?

ilg-ul, 2023-11-14:
@rmyorston, I disabled two variables in the configuration file:

```
sed -i.bak \
       -e 's|^CONFIG_MAKE=y|# CONFIG_MAKE is not set|' \
       -e 's|^CONFIG_PDPMAKE=y|# CONFIG_PDPMAKE is not set|' \
       configs/mingw64_defconfig
```

is this enough?

rmyorston, 2023-11-14:
>is this enough?

Yes, that should be fine.

fanghuaqi, 2023-11-14:
> @fanghuaqi, please try the latest windows-build-tools release, v4.4.1-2.

Hi @ilg-ul , it works well now, thanks for the update.

fanghuaqi, 2023-11-21:
Fixed by windows-build-tools release, v4.4.1-2, close it.

closed 2023-11-21 by fanghuaqi