eclipse-embed-cdt/eclipse-plugins #600: "Interrupt failed" occurred when using GDB OpenOCD Debugging

YuraGakki, 2023-12-27:
### Description

When I used the "GDB OpenOCD Debugging" in the diagram for debugging, I found that the suspend button was not working properly. After pressing it for a period of time, an "Interrupt failed" error was reported. All other debugging functions are normal, such as breakpoints, step over, and so on. When I use C/C++Remote Debugging, this problem will not exist. Additionally, this issue only occurs in Windows and is not present in Linux.

[image]

[image]

From the console and development board, it seems that GDB did not receive the interrupt command. What could be the problem? Has anyone else encountered this problem before?

Thank you in advance for all the replies !

### Versions

* [plug-in version]               6.4.0/6.1.2
* [Eclipse version]              22-09 (4.25.0)
* [Java version]                  JavaSE-17
* [operating system]           Windows 10/11

wang723-stack, 2023-12-29:
This is the log in the. metadata folder when the error occurs.

```
!ENTRY org.eclipse.cdt.dsf 4 10005 2023-12-29 16:03:28.006
!MESSAGE Request for monitor: 'RequestMonitor (org.eclipse.cdt.dsf.debug.ui.actions.DsfSuspendCommand$4$1@6979e769): Status ERROR: org.eclipse.cdt.dsf.gdb code=10004 Interrupt failed.' resulted in an error.
!SUBENTRY 1 org.eclipse.cdt.dsf.gdb 4 10004 2023-12-29 16:03:28.006
!MESSAGE Interrupt failed.
```

From the debug console, it can be seen that GDB did not receive any signal, and there was no response when I manually entered "interrupt" or CTRL+C in the debug console.

[image]

wang723-stack, 2023-12-29:
This is the version information of the GDB I am using.

```
$ riscv32-elf-gdb --version
GNU gdb (2023-06-30_riscv32-elf-298a5116d25) 8.2.50.20190522-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

ilg-ul, 2023-12-29:
Please provide a detailed way to reproduce the issue, using the latest Eclipse, the latest xPack RISC-V GCC toolchain and the latest xPack OpenOCD.

wang723-stack, 2024-01-10:
@ilg-ul Thanks!
### Steps to Reproduce
1. Import a RISC-V C Project.
2. Click the "Debug Configurations" and create a new GDB OpenOCD Debugging.
3. Configure openocd and gdb.
4. Start debugging and "Continue".
5. Press button "Suspend".

When I use the official riscv-none-elf-gdb (xPack), this issue will not occur. So the problem may be that my GDB did not receive an interrupt signal. So could you please tell me how embedded cdt sends an interrupt signal to gdb in Windows? I want to investigate why my gdb cannot receive the signal.

Thank you for any reply!

TommyMurphyTM1234, 2024-01-10:
If you use GDB standalone without any Eclipse/CDT/Embedded CDT support then does Ctrl-C to break into the running program always work OK?

In the scenario that you originally outlined, does increasing the GDB remote timeout before attempting to suspend make any difference? E.g. `set remotetimeout 10`.

* https://sourceware.org/gdb/current/onlinedocs/gdb.html/Remote-Configuration.html

It might make sense to collect and upload a verbose OpenOCD (`openocd -d3`) and GDB traces logs for the issue in case they shed any light on what's going on/wrong here.

closed 2024-03-19 by ilg-ul