eclipse-embed-cdt/eclipse-plugins #607: Error during update: No repository found at http://gnu-mcu-eclipse.netlify.com/v4-neon-updates.

lancos, 2024-06-16:
My PC:
kubuntu 22.04
java openjdk17 (17.0.11)
Eclipse 2024-03 (4.31.0)
Embedded CDT 6.6.0202404031712

Eclipse is working fine, but when i do "Check for updates" it finds updates (version 2024-06), and when i click Next to apply updates it stops with the error:

No repository found at http://gnu-mcu-eclipse.netlify.com/v4-neon-updates.

ilg-ul, 2024-06-16:
This is weird, I thought that references to that URL were updated long time ago when the project was migrated to the Eclipse Foundation.

My suggestion is to reinstall a fresh Eclipse.

TommyMurphyTM1234, 2024-06-16:
In case it's of any use I did this:
```
git clone https://github.com/eclipse-embed-cdt/eclipse-plugins
cd eclipse-plugins
find -type f -print -exec grep "netlify"{} \; 2>&1 | tee grep.txt
```
And in `grep.txt` noticed these results:
```
./scripts/bintray/publish-v4-neon-updates-experimental.command
echo "Install new software from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-experimental"

./scripts/bintray/publish-v4-neon-updates-test.command
echo "Install new software from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-test"
```
But I suspect that they don't actually impact the update mechanism/process and maye simply be redundant vestigial historical references?

ilg-ul, 2024-06-16:
> ... redundant vestigial historical references

that's correct.

lancos, 2024-06-16:
Thanks, just reinstalled fresh new eclipse 2024-06 and it works, no more references to netlify, let's see on the next update in september