eclipse-embed-cdt/eclipse-plugins #609: EmbeddedCDT should compile with SimRel 2024-06

ruspl-afed, 2024-06-27:
### Description

We have one OSGi component [`org.eclipse.embedcdt.internal.packs.ui.MessageConsoles`](https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/master/plugins/org.eclipse.embedcdt.packs.ui/src/org/eclipse/embedcdt/internal/packs/ui/MessageConsoles.java) to let us reduce UI dependencies. The component definition requires `@Component` annotation that was previously contained in `org.eclipse.osgi.services` bundle for many years. Recently [it was decided](https://github.com/eclipse-equinox/equinox/discussions/402) to replace this annotation with its original definition from OSGi.
Unfortunately, target that we use now `https://download.eclipse.org/releases/2020-09/202009161000` does not contain required `org.osgi.service.component.annotations` package.

We need to either upgrade our target or rework OSGi component to extension point, which is much more verbose but less sensitive to Platform changes.

What is your preference @ilg-ul ?

### Steps to Reproduce

1. Try to build with 2024-06

**Expected behaviour:** Build successful

**Actual behaviour:** `org.eclipse.osgi.services` bundle cannot be found

ruspl-afed, 2024-06-27:
> Unfortunately, target that we use now `https://download.eclipse.org/releases/2020-09/202009161000` does not contain required `org.osgi.service.component.annotations` package.

I was wrong, it is able to resolve `org.osgi.service.component.annotations` package, if I neutralize version range for it. Please have a look at #608 @ilg-ul @jonahgraham

ilg-ul, 2024-06-27:
Unfortunately I am not able to comment on these issues.

@jonahgraham, feel free to do whatever necessary to fix them.

jonahgraham, 2024-06-27:
Does this run in 2024-06? i.e. is the problem only building embed-cdt? I ask to know if we need to get a release out soon.

FWIW SimRel 2024-06 and Eclipse 4.32 seem to have org.eclipse.osgi.services. I am not sure what the build problem you are seeing is, but I am fine with your fix. I see that you did other changes with o.e.osgi.services in https://github.com/eclipse-cdt/cdt/commit/88355ecc48cff62d936f3497ca0fd343b42ba0a8

jonahgraham, 2024-06-27:
PS I also expect adding `org.eclipse.osgi.services` to the target platform works around the issue. Regardless, moving our dependencies off the deprecated bundle is a good idea.

PR #608 (2024-06-27): Switch from o.e.osgi.services to org.osgi.service.component.annotations
Follow up for  Equinox change https://github.com/eclipse-equinox/equinox/discussions/402

Fixes #609
closed 2024-06-27 by jonahgraham