eclipse-embed-cdt/eclipse-plugins #610: GD32F303 SVD BUG not show

xzbaijj, 2024-07-06:
[image]

xzbaijj, 2024-07-06:
[image]

TommyMurphyTM1234, 2024-07-06:
You haven't provided any description of the issue and the screenshot image links don't seem to work. 

You need to describe the issue as per the new issue guidelines in order to get useful feedback: 

### Prerequisites

In case you encounter problems while using the plug-ins, please consider the following:

* [ ] read the documentation pages carefully
* [ ] check the [Known issues](http://gnu-mcu-eclipse.github.io/support/known-issues/)
* [ ] check the [FAQ](http://gnu-mcu-eclipse.github.io/support/faq/)
* [ ] check the [GNU MCU Eclipse forum](http://gnu-mcu-eclipse.github.io/support/forum/) for similar problems
* [ ] check the GitHub Issues [tracker](https://github.com/eclipse-embed-cdt/eclipse-plugins/issues/)

If you still could not find a solution, if you have interesting use cases, if you have custom 
configurations, and generally if you have any experience that you want to share with others, 
use the [GNU MCU Eclipse forum](http://gnu-mcu-eclipse.github.io/support/forum/).

If you are convinced that you identified a bug (if you have doubts, use the forum), or you have a 
pertinent suggestion how to enhance the GNU MCU Eclipse plug-ins, continue and register
an issue.

### Description

[Description of the bug or feature]

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behaviour:** [What you expected to happen]

**Actual behaviour:** [What actually happened]

### Versions

* [plug-in version]
* [Eclipse version]
* [Java version]
* [operating system]
* [toolchain version]

Please understand that without being able to reproduce the bug we cannot identify your problem.

---

Note: Remove unnecessary text after reading, before entering your ticket.

TommyMurphyTM1234, 2024-07-08:
Duplicate of this? 

- https://github.com/eclipse-embed-cdt/eclipse-plugins/issues/611

TommyMurphyTM1234, 2024-07-08:
Maybe related to, or the same issue, as mentioned here? 

- https://github.com/leveldown-security/SVD-Loader-Ghidra/issues/13

xzbaijj, 2024-07-09:
Yes, there are spaces at the beginning of the SVD file, which prevents normal use. Can the plug-in automatically handle the spaces at the beginning?

TommyMurphyTM1234, 2024-07-09:
I presume that the plugin caters for SVD files that adhere to the relevant schema:

- https://arm-software.github.io/CMSIS_5/SVD/html/svd_validate_file_pg.html
- https://www.keil.com/pack/doc/CMSIS_Dev/SVD/html/schema_1_2_gr.html

Perhaps the SVD file in question with extraneous (?) spaces at the start of the file is not well formed in terms of that schema?

If that is the case then presumably it's the SVD file and not the plugin that should change (be fixed)?

TommyMurphyTM1234, 2024-07-09:
I don't know where the master/authoritative version of this file lives but this version doesn't seem to have any spaces at the start of the file:

- https://chipselect.org/device_id.php?id=6569&svd_id=0&name=GD32F30x_HD&vendor=GigaDevice

ilg-ul, 2024-07-09:
> Can the plug-in automatically handle the spaces at the beginning?

The plug-in uses standard parsers from the Java library; if your file fails, I would say it is non-conforming and should be fixed.

TommyMurphyTM1234, 2024-07-09:
OK - I guess that the Pack file mentioned here:

- https://github.com/leveldown-security/SVD-Loader-Ghidra/issues/13

namely, this:

- http://gd32mcu.com/data/documents/pack/GigaDevice.GD32F30x_DFP.2.0.0.pack

is the one that is being used and each of the SVD files contained therein do contain extraneous (?) spaces at the start of the files:

- `./SVD/GD32F30x_CL.svd`
- `./GD32F30x_HD.svd`
- `./GD32F30x_XD.svd`

For example:

[image]

I think that the validity or otherwise of these files can be checked using the arm CMSIS-Packs `SVDConv.exe` utility:

- https://www.keil.com/pack/doc/CMSIS_Dev/SVD/html/svd_SVDConv_pg.html

but unfortunately I cannot find how/where to download the CMSIS-Packs utilities and this page doesn't work for me and just says "Loading, please wait..." and never completely loads:

- https://developer.arm.com/embedded/cmsis/cmsis-packs

TommyMurphyTM1234, 2024-07-09:
Ah - found `SVDConv` here:

- https://github.com/Open-CMSIS-Pack/devtools

And running it on the aforementioned SVD files with the recommended `--strict` option generates many `INFO`, several `WARNING` and one `ERROR` for each of these files.

You probably need to flag the issue with GigaDevices (assuming that the relevant CMSIS-Packs file(s) originate from them?) so that they can fix any issues with their files.