eclipse-embed-cdt/eclipse-plugins #614: Is this is a bug?

simonjhy, 2024-08-10:
https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/a99b863f5d6da589751bebfe5334f9b71c8bfdb4/plugins/org.eclipse.embedcdt.debug.gdbjtag.core/src/org/eclipse/embedcdt/debug/gdbjtag/core/dsf/GnuMcuFinalLaunchSequence.java#L101

why in 101 use topToRemove array, but the 102 use **jtagToRemove**  index key and try to remove from orderList. It seems to use topToRemove?

ilg-ul, 2024-08-10:
You are right, this is a bug.

simonjhy, 2024-08-10:
@ilg-ul I want to add a new function that can always get an expression value when use start to debug application. Can you give some advice about it?

ilg-ul, 2024-08-10:
> add a new function that can always get an expression value when use start to debug application

Can you provide more details on this expression and its use case?

simonjhy, 2024-08-10:
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wunused-parameter"
#pragma GCC diagnostic ignored "-Wmissing-declarations"
#pragma GCC diagnostic ignored "-Wreturn-type"

int main(int argc, char* argv[])
{
  // At this stage the system clock should have already been configured
  // at high speed.
	int seconds = 0;
  // Infinite loop
	while (1)
	{
		seconds++;
	}
}
like this code, I want the variable "seconds" will be always shown in a view when I start a debug session.

ilg-ul, 2024-08-10:
Apart the fact that `seconds` will be increased much faster than every second, I think that you should be able to achieve something similar with existing functionality, possibly with some elaborate watch expressions.

Unfortunately I don't remember details now...

ilg-ul, 2024-08-10:
Please do not add further comment not related to the issue subject.

ilg-ul, 2024-08-27:
Fixed in 6.6.1.

closed 2024-08-27 by ilg-ul