eclipse-embed-cdt/eclipse-plugins #617: In version 2024.06, there are cases of garbled Chinese characters when processing them in the console.

laomaolaile, 2024-09-23:
When compiling a project, a .bat file is added to the Pre-build steps. However, when an error occurs while executing the .bat file, the error message is in Chinese, but these Chinese characters are displayed as garbled text in the console.

[image]

However, in the previous version, this function worked normally.

[image]

 Is there any temporary solution to this problem, or can it be fixed through some settings so that it can display properly?

TommyMurphyTM1234, 2024-09-23:
What is the console encoding set to in each case? 

_Run > Run Configurations > Common Tab > Encoding_ 

If it's not UTF-8 then set it to that and try again.

laomaolaile, 2024-09-23:
This doesn't seem to have anything to do with run, it appeared during the build project.

[image]

TommyMurphyTM1234, 2024-09-23:
Did you check what I suggested anyway?

laomaolaile, 2024-09-23:
I compared the console settings in two versions and they are both the same. I think this issue is not a configuration problem.
Is it possible that there is no compatibility with Chinese when processing the output stream

laomaolaile, 2024-09-29:
@ilg-ul Excuse me, do you have a solution to this problem?

ilg-ul, 2024-09-29:
Unfortunately I don't know where the problem comes from, but my first guess is Eclipse itself.

Could you try a project in a standard CDT project, i.e. not embedded?

laomaolaile, 2024-09-29:
@ilg-ul I try a project in eclipse-cpp-2024-03-R-win32-x86_64   It is normal, but in i try a project ineclipse-cpp-2024-06-R-win32-x86_64, garbled code is reported, as you guessed. Thank you.

ilg-ul, 2024-09-29:
@jonahgraham, any suggestion?

TommyMurphyTM1234, 2024-09-29:
Might this change in 4.32 have something to do with it perhaps? 

- https://eclipse.dev/eclipse/development/readme_eclipse_4.32.php

> The Eclipse SDK supports GB 18030 (level 1), the Chinese code page standard, on Windows, Linux and the Macintosh.

ilg-ul, 2024-09-29:
@laomaolaile, are you sure that the texts used by your project are encoded using the standard GB 18030?

laomaolaile, 2024-09-30:
my project is also a template project created in Eclipse.

[image]

[image]

From this image, it can be seen that the error is actually Windows output, prompting "please press any key to continue"

TommyMurphyTM1234, 2024-09-30:
@laomaolaile, what about these two points/questions? 

> @laomaolaile, are you sure that the texts used by your project are encoded using the standard GB 18030?

And...

> Maybe something needs to be done to enable that code page in Windows or for the Eclipse a Terminal so that the text can be displayed correctly?
>
> Can it be changed here perhaps?
> 
> - https://help.eclipse.org/latest/index.jsp?topic=%2Forg.eclipse.pde.doc.user%2Fguide%2Ftools%2Flaunchers%2Fcommon.htm

ilg-ul, 2024-09-30:
> the error is actually Windows output, prompting "please press any key to continue"

well, in this case the encoding known by Windows is probably different from the actual encoding used by the Eclipse terminal.

jonahgraham, 2024-10-04:
I cannot reproduce this problem. I installed Chinese (Simplified, China) language pack and made it the Windows display language, but that didn't change any behaviour. `pause` still displayed in English.

@laomaolaile do you have any additional insights on how I can reproduce this problem to see if we can fix this regression.

ilg-ul, 2024-10-04:
My experience with Windows is very limited, but I would say that somewhere in the Windows settings there should be a choice for the code page, possibly reflected in an environment variables, or visible with a tool (systemview?).

Similarly the Eclipse terminal should have a setting for the code page.

Obviously these two should match.

laomaolaile, 2024-10-12:
I provided my test files, I don't know if this can reproduce the problem. @jonahgraham 

[image]

```
echo "我是中文";
```

[copy.zip](https://github.com/user-attachments/files/17349199/copy.zip)

ilg-ul, 2024-10-12:
Once again, the code page settings you have in your Windows configuration does not match the settings you have in Eclipse.

There is nothing the Eclipse Embedded CDT can do to fix your configuration.

closed 2024-10-12 by ilg-ul

laomaolaile, 2024-10-12:
This is clearly a bug, as Eclipse did not handle the Chinese language properly when dealing with the return of make. I don't think this has anything to do with configuration.  @ilg-ul 

[image]

ilg-ul, 2024-10-12:
> This is clearly a bug

The Eclipse Embedded CDT does not handle the terminals itself, it inherits them from Eclipse CDT/Eclipse Platform.

Please report the bug there.