eclipse-embed-cdt/eclipse-plugins #619: Eclipse Lock up when debugging with Jlink

ilg-ul, 2024-10-05:
### Discussed in https://github.com/eclipse-embed-cdt/eclipse-plugins/discussions/618

<div type='discussions-op-text'>

<sup>Originally posted by **trampas** October  4, 2024</sup>
I have been having long term issue where Eclipse locks up when debugging with Jlink.  Read here for more detail: https://github.com/eclipse-cdt/cdt/discussions/906#discussioncomment-10845101</div>

ilg-ul, 2024-10-05:
There are quite a lot of exceptions in the logs, this is probably a bug.

I'll try to investigate, but it might take some time.

SnowyTulip, 2024-12-08:
Hello, Liviu Ionescu,

I am very interested in this issue and have been working to reproduce it.

**Here is my setup:**

* SEGGER J-Link GDB Server V7.96g
* Windows 11
* STM32F407 development board
* J-Link V9 (SWD interface)
* Debug version of Eclipse Embed-CDT built from the source code in this repository

**The steps** to reproduce the issue are as follows:

1. Connect the J-Link to the STM32 development board.
2. Create a new LED template project using Embed-CDT.
3. Set up a new J-Link debugging configuration and start debugging.
4. After hitting a breakpoint, unplug the J-Link (please note that this is not the same as disconnecting the power supply to the development board. Based on my testing, removing the power supply does not reproduce the issue).

**Result**
The IDE suddenly becomes extremely slow and eventually freezes without reporting any errors.
[eclipse_error.log](https://github.com/user-attachments/files/18053393/eclipse_error.log)

The debug log I captured is almost identical to the one shared here. Strangely, even though the freeze in Eclipse is caused by J-Link, there are almost no J-Link-related exceptions in the log. Instead, there are many exceptions related to SWT handles and the UI event loop.

### Solution

Since the issue seems related to the UI events, I noticed that there appears to be an anomaly in J-Link's semi-hosting mode, where there are many unrecognized characters being printed. The issue seems to stem from here! The semi-hosting thread continuously printing data likely consumes a large number of SWT handles. As a result, the UI is not actually frozen; instead, the excessive printing in semi-hosting mode consumes so many resources that Eclipse runs very slowly.

To verify my hypothesis, I modified the input/output stream part of the SemihostingProcess [in the code](https://github.com/eclipse-embed-cdt/eclipse-plugins/blob/master/plugins/org.eclipse.embedcdt.debug.gdbjtag.jlink.core/src/org/eclipse/embedcdt/debug/gdbjtag/jlink/core/SemihostingProcess.java#L137C1-L137C38). I added a 1-second delay each time it printed to the console. After making this change, when I repeated the steps above, the IDE no longer froze! (In fact, I also tried disabling the console output, and the result was the same.)
[image]
[image]

ilg-ul, 2024-12-08:
Frankly, the Eclipse Embedded GDB code is very old and I have difficulties to follow the details. We'll need someone from the Eclipse CDT project to help us.