eclipse-embed-cdt/eclipse-plugins #621: Eclipse embedded v2024-12 starter.exe problem on Win10

vladi552, 2025-01-02:
### Description
When I compile STM32 project and compilation time is more then approx 5sec, the unwanted starter.exe window appears on the screen and automatically closed when compilation is finished.

### Steps to Reproduce
Not needed any special, just create STM32 project where compilation time is to long.

**Expected behaviour:** [What you expected to happen]
This starter.exe window not needed to appear.
**Actual behaviour:** [What actually happened]
The starter.exe window cannot be avoided in display (resize not possible, pozition also rized...)
### Versions
Win10 IoT LTSC, Eclipse 12-2024
[plug-in version]: org.eclipse.cdt.core.win32.x86_64_11.6.1.202407022008

[image]

ilg-ul, 2025-01-02:
Thank you @vladi552 for your bug report.

Is this issue specific to 2024-12? Could you try the same sequence of steps with 2024-09 or earlier?

jonahgraham, 2025-01-02:
This is probably strictly a CDT problem, rather than embedded CDT.

> starter.exe window appears on the screen and automatically closed when compilation is finished.

Besides the window appearing, is there any other behaviour issues you see?

Which version of Windows 10 are you using (I see _IoT LTSC_ in your report, but no version number). Do you know if you have any special security policies in place on the enterprise machine, such as preventing CDT from creating hidden windows?

> Is this issue specific to 2024-12? Could you try the same sequence of steps with 2024-09 or earlier?

I was going to ask something similar, is this a regression you have observed from any previous version?

vladi552, 2025-01-02:
> Which version of Windows 10 are you using
Windows version: 21H2 19044.5247

> Could you try the same sequence of steps with 2024-09 or earlier?
Ok. Need a bit of time... I reported later.

> Besides the window appearing, is there any other behaviour issues you see?
No. At the moment all other behaviours is OK in 2024-12.

> Which version of Windows 10 are you using
Windows version 21H2 19044.5247

jonahgraham, 2025-01-02:
> Windows version: 21H2 19044.5247

That is the most recent build of Windows 10, released in Dec 2024. I haven't seen a report like this before and wonder if there has been a regression on Windows side. I don't have easy access to Windows 10, but I will try to update my Windows 11 test machine to the December build (26100.2605) to see if that exposes the issue.

>> Could you try the same sequence of steps with 2024-09 or earlier?

> Ok. Need a bit of time... I report it later.

I would be very surprised if 2024-09 makes any difference as the version of CDT is the same in that version.

Have you been using Embedded CDT for long, or are you a new user. If you have been using Embedded CDT for a while on this Windows Enterprise machine, have you ever seen this starter.exe before? i.e. is this a regression in behaviour, or new to you behaviour?

vladi552, 2025-01-02:
2012-09 have the same starter.exe problem. Clear install, clear workspace, I just import my project into workspace and compile...
I have a long pause in development, so my latest Eclipse has been a few years ago. However, I never seen this starter.exe problem. Some weeks ago I decide to remove/downgrade this stupid buggy Win11 and have a rest with Win10 IoT LTSC which is working great for me - quickly and no any problems. I now really very happy with it and haven't win11 "updates" anymore. But problem with Eclipse arrived...

TommyMurphyTM1234, 2025-01-02:
Does the `starter.exe` window appear for any other tasks that involve Eclipse/CDT launching external tools such as the compiler, debugger, user specified external tool launcher, etc.? Or is it only in this specific scenario?

vladi552, 2025-01-02:
At the moment my investigation result: starter.exe does not appears with the AVR projects. Trying futher...

vladi552, 2025-01-03:
Catch the problem point. It's ARM toolchain related:
1. [ARM tools](https://disk.yandex.ru/d/4J88wBkrhcmAIA) with starter.exe problem in Eclipse.
2.  [ARM tools](https://disk.yandex.ru/d/uk2gEFb63YHdxw) without starter.exe problem in Eclipse.

jonahgraham, 2025-01-03:
Thanks @vladi552 - that is some great input. Can you explain what the two links are, like how I reproduce their content? Did you compile these yourself, or download them from somewhere?

jonahgraham, 2025-01-03:
I have reproduced the problem on my computer. Once I figure out more I will post additional details.

jonahgraham, 2025-01-03:
:-( this is weird. It is something to do with combinations of tools, but not sure exactly what.

If I create a Makefile in a plain makefile project with these contents:

```
all:
	arm-none-eabi-size  --version
	#arm-none-eabi-size  --format=berkeley "blinky.elf" 
	ping 192.0.3.2 -n 1 -w 1000000
```
(the ping is to simulate sleep)
and run it, I get the starter.exe window.

If I delete the *commented out* line, I don't get the starter window.

This is using make `xPacks\@xpack-dev-tools\windows-build-tools\4.4.1-3.1` and `xPacks\@xpack-dev-tools\arm-none-eabi-gcc\13.3.1-1.1.1` (these are the versions I happened to have installed for something else. I can/will try with other versions.

jonahgraham, 2025-01-06:
This is an annoying issue, but as it is primarily cosmetic it is lower down on my todo list. If you observe any errors as a result of this I can raise the priority.