eclipse-embed-cdt/eclipse-plugins #625: When OpenOCD starts up slowly or abnormally, the IDE will report an error.

laomaolaile, 2025-02-10:
When OpenOCD starts up slowly or abnormally, the IDE will report an error.

[image]

[image]

Is it possible to consider allowing the OpenOCD startup timeout period to be configurable by the user?

[image]

ilg-ul, 2025-02-10:
This might be a possibility, but I'd be also interested to know the reason why openocd takes so long to start.

TommyMurphyTM1234, 2025-02-10:
Similar issue here? 

- https://github.com/eclipse-embed-cdt/eclipse-plugins/issues/585

djanovy, 2025-02-10:
Just a guess, but would adding the `set remotetimeout` work?

[image]

ilg-ul, 2025-02-10:
I'm not an openocd user and I don't have any experience with these timeouts, so I don't know if this configurable timeout is helpful or not.

If you do, please suggest a PR and I'll consider it.

TommyMurphyTM1234, 2025-02-10:
> Just a guess, but would adding the `set remotetimeout` work?
> 
> [image]

I don't think so. That helps when OpenOCD is up and running but, for some reason, slow to respond to GDB Remote Serial Protocol commands and needs more time before timeouts much in.

What this issue is referring to is where Embedded CDT is trying to launch OpenOCD in the first place but, for some reason, it takes "too long" to launch, and Embedded CDT times out, gives up and reports an error. 

I have no idea what the timeout used here is, if it's some general timeout in Eclipse/CDT or specific to the Embedded CDT OpenOCD launch support, and if it's configurable via `eclipse.ini` or some other configuration file even if not via the GUI.

laomaolaile, 2025-02-11:
My scenario is Eclipse installed on a workstation, and there may be some software interception in between, such as antivirus software. In this case, there may be a timeout situation. Or openocd could have been directly killed