eclipse-embed-cdt/eclipse-plugins #629: CMSIS Packs, unable to download

bureaua, 2025-04-28:
By pressing Refresh button on "CMSIS Packs" tabs, by using the original URL: https://www.keil.com/pack/index.pidx -- it does not work

```
2025-04-28 10:07:13
Update packs job started.
Parsing "https://www.keil.com/pack/index.pidx"...
javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
java.io.IOException: The system cannot find the path specified
Update packs completed in 0s.
```

After changing the URL to https://sadevicepacksdqaus.blob.core.windows.net/idxfile/index.pidx -- started to work as expected

But when its finished -- some of the packs were not downloaded due to the same issue:

```
javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target https://www.keil.com/pack/Keil.STM32F0xx_DFP.pdsc
Missing "c:\Users\Name\AppData\Roaming\CMSIS-Packs\.cache\Keil.STM32F0xx_DFP.3.0.0.pdsc", ignored by user request.
```

By opening https://www.keil.com/pack/Keil.STM32F0xx_DFP.pdsc --  I can download the PDSC file

ilg-ul, 2025-04-28:
Probably the Keil servers updated their keys to versions incompatible with the Java code used by Eclipse.

Anyway, the CMSIS Packs plug-in is long deprecated. No longer recommended for new projects.

bureaua, 2025-04-28:
As I understood, to be able to observe the status of the MCU peripheral registers, I have to download CMSIS Packs and select MCU that I am using in the project settings

[image]

As it is "No longer recommended", could you please point me to which solution I have to use instead... ?

ilg-ul, 2025-04-28:
> could you please point me to which solution I have to use instead

download the SVD file manually, copy it to the project, and configure the debug plug-in to use it.