eclipse-embed-cdt/eclipse-plugins #633: The Command History feature in the Debugger Console is not working as expected.

laomaolaile, 2025-05-15:
According to the official Eclipse documentation, the Debugger Console supports a Command History feature. 

https://wiki.eclipse.org/CDT/Archive/FullGDBConsole

[image]

However, when debugging a RISC-V project in Embedded CDT (embed-cdt), I found that the Command History does not function properly.

Interestingly, when I created a regular C project in the same environment and used the Debugger Console, the Command History worked normally. I'm wondering which component or configuration might be causing this issue in the RISC-V project context.Any insights or suggestions would be greatly appreciated.

[image]

[image]
﻿
Thank you for your attention to this matter!

ilg-ul, 2025-05-15:
It might be possible that the features related to Debug console were added to CDT after the EmbedCDT was implemented (very long time ago!), and require some additional setting to be enabled.

@jonahgraham, do you know how to enable this in EmbedCDT?

jonahgraham, 2025-05-15:
I think org.eclipse.embedcdt.debug.gdbjtag.core.dsf.GnuMcuGdbBackend and its subclass GdbBackend are copied from an old version of the org.eclipse.cdt.dsf.gdb.service.GDBBackend class hierarchy. The newer org.eclipse.cdt.dsf.gdb.service.GDBBackend_7_12 has the needed functionality. If it is possible to make GnuMcuGdbBackend extend GDBBackend_7_12 then it should just work™

PS This won't work on Windows - I can't remember the state on mac, but I would guess it should work there as mac has a real PTY implementation.

ilg-ul, 2025-05-16:
If I remember right, the reason for copying that class locally was that I could not extend it to add the needed functionality.

If the new class is more configurable and I can extend it, we can consider extending it.

However, this will probably also require raising the Eclipse/CDT base version, which will take some efforts.

fanghuaqi, 2025-09-10:
Dear @ilg-ul, I would like to inquire if there have been any developments regarding this issue. Specifically, will this feature be incorporated in the future?

ilg-ul, 2025-09-10:
Unfortunately new developments in the debug plugins require more efforts than my limited resources can provide.