eclipse-embed-cdt/eclipse-plugins #643: External Builder: Cannot run program "rm": Command line too long

fanghuaqi, 2025-09-25:
Hello, I have a big project which have a lot of files, and I faced a issue during link and clean process, it is caused by windows 32k command line limit, I have searched in website, found the following bugs

- related to long rm command line https://bugs.eclipse.org/bugs/show_bug.cgi?id=505882
- related to long link command line https://bugs.eclipse.org/bugs/show_bug.cgi?id=72965

For the first one, is current embedded cdt also fixed, for our IDE based on embedded cdt, this issue still present, but I can manually change  the  `C/C++ Build -> Tool Chain Editor -> Current builder -> Gnu Make Builder` select it and do Apply and close, and then do clean project and build project, and clean project, then you will see the clean target changed.

~~~
# orginal target 
clean:
	-$(RM) $(CCM_DEPS)$(C_UPPER_DEPS)$(SECONDARY_LIST)$(SECONDARY_SIZE)$(CXXM_DEPS)$(C_DEPS)$(CC_DEPS)$(SX_DEPS)$(C++_DEPS)$(OBJS)$(CXX_DEPS)$(SECONDARY_FLASH)$(ASM_DEPS)$(S_UPPER_DEPS)$(C++M_DEPS)$(CPP_DEPS) 2222.elf
	-@echo ' '
# new target
clean:
	-$(RM) 2222.elf 2222.hex 2222.lst 2222.siz
	-@echo ' '
~~~

I have create a project to test it, you can check it in [ide_better_clean-target.zip](https://github.com/user-attachments/files/22527643/ide_better_clean-target.zip), this is a git maintained project, so just check last commit change via `git log -1 -p`

Regarding the link command line too long issue, we can try with gcc response file like this

We can add new line in elf generation target like this

~~~shell
@echo 'Building target: $@'
# Add after it
@echo 'Generate response file: $@.linkargs'
$(file >$@.linkargs,$(OBJS) $(USER_OBJS) $(LIBS))
# replace key words
In later lines replace $(OBJS) $(USER_OBJS) $(LIBS) to @$@.linkargs
~~~

Then it can fix linker command line too long issue, but can this be done directly in embedded cdt?

Thanks
Huaqi

ilg-ul, 2025-09-27:
For compatibility reasons, ECDT is based on slightly older code.

If the fixes are in the code that ECDT inherits, yes; if the changes are in more recent code, no.

ilg-ul, 2025-09-29:
> hope this old issue could be fixed soon

Yeah, good luck with this.